earlephilhower/ESP8266Audio #86: AudioOutputI2S not works.

rkoptev, 2018-04-24:
Hi!

I'm using ESP8266 and PCM5102 DAC and want to play audio on it. However, examples not working for me. I hear no sound, just quiet square noise sometimes (~50Hz). I tried PlayMP3FromSPIFFS replacing AudioOutputI2SNoDac with AudioOutputI2S, also I tried PlayRTTTLToI2SDAC but both ended up with nothing. There are no any problems in serial port:
`RTTTL start`
`RTTTL done`
`RTTTL done`
`RTTTL done`

The connection is as follows:
BCK - GPIO15 (D8 on NodeMCU)
DIN - GPIO3 (RX on NodeMCU)
LCK - GPIO2 (D4 on NodeMCU)

Here also the photo of my setup:
[image]

earlephilhower, 2018-04-24:
I have that same board, and there are several pins you need to tie off in addition to just wiring in the clocks and data.  Also, it's hard to see in the photo but it looks like not all the the VCC, 3.3V, and GND pins are wired up.  All inputs should be tied high or low as appropriate, I don't think there are any pullup/downs on that board.

Check the chip datasheet for more info:
http://www.ti.com/lit/ds/symlink/pcm5102.pdf

rkoptev, 2018-04-24:
As you suggested, I tied some pins to GND or 3.3V:
XMT (mute) - to high (disable mute)
FMT (data format) - to low (I2S)
SCL (system clock) - unwired (to use internal clock)
DMP (De-emphasis control for 44.1kHz sampling rate) - to low (to disable)
FLT (Filter select) - to low (normal latency)

but nothing changed

earlephilhower, 2018-04-25:
You've got a wiring or power supply problem, or the chip is dead, then.  Here's my wiring:
3.3V from ESP8266 -> VCC, 33V, XMT
GND from ESP8266 -> GND, FLT, DMP, FMT, SCL
Then BCLK->BCK, I2SO->DIN, and LRCLK(WS)->LCK

rkoptev, 2018-04-25:
I wired module as you described. The VCC voltage is 3.22V which is fully OK for this chip (3.0-3.6V normal operating voltage according to datasheet). I investigated the signals with an oscilloscope and found 1.655MHz on LCK pin, 51.9 kHz on BCK pin and some data in sync with LCK on DIN pin.
I'm already completely confused with these names of pins. But I assumed that these clock signals are confused. And my assumption was correct. After I changed BCK and LCK pins I hear the music!!

So, if someone will use PCM5102 in conjunction with NodeMCU chip, notice the working connection scheme:

VCC - 3.3V
3.3V - 3.3V
GND - GND
FLT - GND
DMP - GND
SCL - GND
BCK - D8 (GPIO 15)
DIN - RX (GPIO 3)
LCK - D4 (GPIO 2)
FMT - GND
XMT - 3.3V 

Thank you, @earlephilhower for you help!

closed 2018-04-25 by rkoptev

rkoptev, 2018-04-26:
In my project I'm using PCM5102 based module with 3.5mm jack. There is also version with RCA connectors on this particular chip. They mostly used for Raspberry Pi, but actually they are very decent quality (32bit@384kHz) low price audio DAC.

Here is the link on AliExpress: https://www.aliexpress.com/item/PCM5102-Module-Audio-Board-With-Stereo-Jack-Beyond-ES9023-PCM1794A-DAC-For-Raspberry-Pi-pHAT-Sound/32846650310.html

Happy to help you in case of any other questions or if you will want to test some staff.

And thank you so much for such a great work! This lib helped me a lot.

rkoptev, 2018-04-26:
And yes, I think, it's very good idea to have pinouts explained on readme, because I spent a significant amount of hours trying to understand what the problem with wiring.