earlephilhower/ESP8266Audio #87: Compatibility issue with Mesh library

rkoptev, 2018-04-26:
Hi again,

I'm having problems working with this library together with painlessMesh library: https://github.com/Coopdis/easyMesh

With this library I receive no sound and no errors from ESP8266Audio. It looks like i2s interface is starting, the clock signals are working normally. However, the data line is empty. Does the library use any global resources, such as timers or interrupts, that can conflict with other libraries?

And if so, can it be fixed on ESP?

Thank you in advance,

Ruslan

earlephilhower, 2018-04-26:
Howdy,

The ESP8266Audio libs don't use HW resources other than the I2S hardware.  Make sure you're calling the wav->loop() method frequently in your loop(), that there's no busy-waiting in the lib you're trying to use (if it's busy-waiting, add in calls to the audio->loop() method or else the I2S will underflow), etc.

Also, obviously, make sure that you've got enough free RAM to handle whatever you're doing.  The MP3 lib eats a ton of memory, so you may want to start with the RTTTL libs which use next to none, just to verify your chain.

Good luck
-EFP3

rkoptev, 2018-04-27:
Thank you for the explanation, @earlephilhower,
Looks like the issue was really in free memory. I have only 20 kilobytes of free heap, and seems, it' not enough for MP3. The RTTTL works fine. So I replaced MP3 with WAV and all is working now. Of course, I had to lose compression, but I think I'll fit in 3MB without problems)

closed 2018-04-27 by rkoptev