earlephilhower/ESP8266Audio #89: Using 2N3904 does not make any sound

abdollahpour, 2018-05-02:
I use Adafruit mono Amplifier and it works well. But using MMBT3904LT1G transistor does not work at all. It does not play anything even noise. If I connect RxD directly to the speaker it works (very low) that shows things are right in my app. I use 8Ω 0.5W speaker.

msmcmickey, 2018-05-02:
Things to check:

1. Is the transistor connected properly? Check the datasheet for this package style and make sure you have the leads connected properly. This package has three leads, and the lead that is by itself in the middle of the one side is the collector, not the base as you might expect it to be.

2. If connected properly, do you have ~5 volts between the collector and emitter?

3. Was the transistor possibly damaged/overheated during soldering, or by connecting it improperly? Out-of-circuit diode check voltage drop test using a multimeter from base->emitter and base->collector should be between .5 and .7 volts. If it's shorted or open or conducting in both directions, then replace it and make sure it's connected properly. 

This is a very simple circuit and it works. If you have measurable output from the 8266, everything is connected properly, and you have a working transistor, it will work as described.

earlephilhower, 2018-05-03:
Thanks, @msmcmickey.  You've covered it very well!

msmcmickey, 2018-05-04:
Thank you, @earlephilhower. Your library is brilliant. I use the 8266's for a lot of things, mostly temperature sensing, aquarium light controls, modem/access point resets, etc. and having radio as a major hobby (as well as being a former engineer), I thought about trying to use them for audio but honestly didn't think they'd have the processing power necessary. You really nailed this perfectly. I approached this with amusement at first, but once I got one programmed and heard fairly stable and clean audio coming from the RX pin of an 8266, I was hooked! What a great project. Thank you so much.

earlephilhower, 2018-05-15:
I've added your tips to the readme to preserve them for posterity.  Thanks again!

closed 2018-05-15 by earlephilhower

abdollahpour, 2018-05-23:
I test it again with another transistor. It works I think the problem with overheated maybe. I could say it's better with the amplifier I use 0.5W 8 Ω speaker it's not loud enough but it amazingly works. Great job. Thanks