earlephilhower/ESP8266Audio #90: Intermittent buzzing sound while running WebRadio example on ESP-32

balimansky, 2018-05-02:
All - I am thrilled to have discovered these Audio libraries as I am starting to build and tinker with the ESP32.  I have successfully been able to run the Web Radio example on my ESP32 Dev Board (generic), however I've come across an interesting issue.

In my current simplistic set up I have the demo running with an external PCM-5102 I2S DAC decoder but have also noticed the same issue when I use the build in DAC connected directly to a powered speaker.

The issue:

I’ve noticed a frequent and intermittent low frequency buzz as the music is streaming/playing.  It sounds like maybe a buffering sort of issue but I’m not sure how to tinker with the settings to experiment with this. 

Another possibility (that I was considering) has to do with CPU interrupts or perhaps some sort of clock frequency issue?

Do any thoughts come to mind on how to potentially address this?

Thanks!
Ben

earlephilhower, 2018-05-03:
Howdy,

It could be buffer underrun, but normally on the ESP32 that manifests itself as "stuttering."  On the ESP8266 when an underflow occurs the buffers are zeroed out causing abrupt, short silences, but the 32 doesn't do this so you end up replaying sounds from ~.2-.5 seconds ago.

Why not try the RTTTL example and see if the same buzzing occurs there?  That is a very low CPU usage generator so underflows are very, very hard to cause.

If RTTTL has buzzing, then, it's electrical and related to something in the amp, wiring, power supply, etc.

If RTTTL is clear, then it may be underflow.  You can adjust the buffer sizes in the example code upwards on the ESP32.  You could also try a different URL to ensure it's not a network BW issue (not your internet connection, but either the WiFi link to the 32 or over the public internet).

Good luck!

balimansky, 2018-05-06:
Hi Earle (and others) -

I wanted to report some fascinating observations and also that my current issues have been SOLVED!  Hopefully this information and experience will benefit others, as this has been a terrific learning experience.

First and foremost I want to thank Earle and the community here for these libraries.  I am thrilled to be able to leverage them as the foundation for some cool audio webradio streaming I'm been doing.

In terms of the intermittent buzzing that I had been experiencing there are a few components of the solution I found:

1. As Earle suggested, I found the above test (RTTTL) to be a critical part of finding a solution.  When I ran the simple test I had no issues, which gave me confidence that the problem was not related the basic hardware set up I had built.

2. I refocused on the web radio, and tried different internet radio sources, thinking the issue might be related to the quality of the source.  This didn't seem to correlate to the problem - the buzzing noise was still apparent.

3. As I was tinkering with the PCM5102 I2S decoder I found that by accidentally touching some of the leads such as FLT and FMT some of the buzzing was eliminated.  I found other documentation on this specific board I was using and noted that these two pins needed to be set as follows:  FMT=low for I2S format and FLT=low for normal latency for filter select.

4. Next, and this is really interesting, I had an "environmental" issue.  The workspace where I do my development is a steel metal table with a thin piece of disposable paper on top to keep the surface clean.  My ESP32 dev board is attached to a solder less breadboard which I placed on top of my table.  Apparently the immediate proximity of the breadboard to the table (i.e., sitting directly on top of it) causes RF interference - enough so that the audio output had a lot of static.  I discovered that if I lifted the board off the table by a small amount the problem went away.  I don't have enough RF experience to know exactly what was going on but am guessing that a magnetic field was created, enough so to disrupt the sensitive I2S output from the ESP32.  **Any thoughts here would be appreciated.**

5. By this point I had really improved the audio output but I still had the original problem - this intermittent weird feedback of gapping in the audio stream.  Given Earle's original suggestion of using a simple example, I found this to be critical - simply simply simply.  I went back to the web radio code and removed the client side of the web server.  My goal is to provide my user with a static list of internet radio IP addresses (essentially a list of pre-programmed radio stations) so I didn't really need the server code.  In doing that I stripped down the code within the Arduino loop function and this appears to have solved this strange buffering problem.

I completely understand that this isn't really a "solution" as I'd need to better determine the cause, but I'm guessing that there was something that was causing either an interrupt or delay which interfered with proper I2S output/streaming.

**Any thoughts on this would be appreciated!**

Thank you all!

Ben

earlephilhower, 2018-05-06:
Good debugging!

I assumed everyone would know you don't leave inputs floating, but this is the 2nd time someone using that board (same as I have wire-wrapped now, actually) was missing the non-i2s connection wiring.

My own setup is as such:
> 3.3V from ESP8266 -> VCC, 33V, XMT
> GND from ESP8266 -> GND, FLT, DMP, FMT, SCL
> Then BCLK->BCK, I2SO->DIN, and LRCLK(WS)->LCK

As for your 4. you could try shorter leads or direct-connect (i.e. wirewrap or solder) for the I2S signals and parallel Vdd and GND wires with maybe a decoupling cap near the i2s board.  The clock going in is at 32x the output frequency (~1.4MHz) which is definitely in the range of interesting RF effects if there's cap or resistance issues (like in a solderless breadboard).

The no. 5 this is simply a timing issue.  If you sprinkle add'l mp3->loop() calls around the web interface that may fix it.  On the 8266 it's fine except for when a web connection is made which causes a hiccup (only 1 core).