earlephilhower/ESP8266Audio #91: Application crashes on new AudioOutputI2SNoDAC()

abdollahpour, 2018-05-04:
I have a problem with I2S. Here is the code:

    AudioGeneratorWAV *wav;
    AudioFileSourceSPIFFS *file;
    AudioOutputI2SNoDAC *out;

    void setupAudio() {
        SPIFFS.begin();
        out = new AudioOutputI2SNoDAC();   // ------------ Error is here ------------------
        wav = new AudioGeneratorWAV();
    }

    void audioLoop() {
        if (wav->isRunning()) {
            if (!wav->loop()) wav->stop();
        }
    }

    void audioPlay() {
        file = new AudioFileSourceSPIFFS("/file.wav");
        if (!wav->isRunning()) {
            wav->begin(file, out);
        }
    }

I use [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) for TFT display also.

* ESP8266 Firmware: 2.4.1
* TFT_eSPI: Last version
* ESP8266Audio: Last version

The application crashes on the line I marked (AudioOutputI2SNoDAC). It does not show any error or something, but ESP goes crazy and even TFT print some mess pixels.
The weird part is the code worked before I update ESP8266Audio, TST_eSPI and the firmware version 2.4.1. Unfortunately, I did not make a backup and I have no idea what version worked before.
ESP8266Audio & TFT_eSPI work speratly

earlephilhower, 2018-05-04:
Without a full failing code example or a decoded stack trace, I really can't help you since it seems each works fine w/o the other.

My guess is that you're running out of RAM and the I2S is failing a malloc somewhere.  You can dump the ESP.getFreeHeap() value just before the call to verify this.  If it's only a few KB, that would explain things.  In that case, there's not much that can be done.

Other things may be something to do with the HW, maybe the two libs are trying to use the same I2S driver to send pixel and audio data (like some folks use I2S to control steppers or neopixels).  The audio lib doesn't need anything other than I2S and RAM to work.

earlephilhower, 2018-05-04:
You could also try GIT HEAD of the 2.4.1 Arduino IDE.  I rewrote the I2S to support input, too, and added some checks for OOM (it won't play if no RAM, but it also shouldn't crash).

JosefKuchar, 2018-05-13:
I had same problem but with AudioOutputI2SDAC. AFAIK when `SPIFFS.begin();` is before `new AudioOutputI2SDAC();` it randomly crashes...

earlephilhower, 2018-05-15:
Would either of you please get a decoded crash dump and sample source file and post them?

If I can reproduce it locally, I can take a look at it.  Without that, I'm not really able to help much as I've never seen this problem.

earlephilhower, 2020-08-08:
There was a fix to the code a couple years ago related to the I2SNoDAC var init, fixing this.

closed 2020-08-08 by earlephilhower