earlephilhower/ESP8266Audio #92: Seeking when using AudioFileSourceHTTPStream

HumamHelfawi, 2018-05-04:
I am using AudioFileSourceHTTPStream to stream an Mp3 file from a webserver. However, I noticed that the seek function is not implemented in this class. 

Is there any trick to seek while using AudioFileSourceHTTPStream ?

earlephilhower, 2018-05-04:
There's not an option to do that now.  HTTP has nothing like a seek capability.  You sent a GET and get bytes one after the other until the EOF.  There is not enough memory to buffer the entire file, of course, so you can't cheat that way.

You could drop the existing HTTP connection and make a new one with a `range: xxxx-` header on a seek, but connection setup time on the ESP8266 is rather long.

If you do hack something like this up, Id be happy to see a merge request!

HumamHelfawi, 2018-05-04:
Thanks for your fast reply and for the great library in general!
Nothing in my mind currently.. I will post it indeed if I found someway