earlephilhower/ESP8266Audio #94: MOD playing on DAC - awful sound (TDA1543-TDA1387)

Airrr17, 2018-05-13:
Hi!
When I using webradio (mp3) with 100% volume via webinterface, I hear really bad sound, but once i switch volume to 40% it becomes very good!
May be it will give you a clue. Now to the problem:

When playing MOD files it's always awful sound (especially bass) regardless volume. 
out->SetGain (xxx); doesn't work properly. it do makes output sound quieter but quality remains
the same. 

It goes even worse with MOD files bigger then 250-300kb.

its slightly modified code for sd card.

#include <Arduino.h>
#include "AudioGeneratorMOD.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include <ESP8266WiFi.h>
#include <SPI.h>
#include <SD.h>
AudioGeneratorMOD *mod;
AudioFileSourceSD *file;
AudioOutputI2S *out;

  file = new AudioFileSourceSD("/TRANZE~1.MOD");

  out = new AudioOutputI2S();
  mod = new AudioGeneratorMOD();
  mod->SetBufferSize(3*1024);    //3
  mod->SetSampleRate(44100);
  mod->SetStereoSeparation(64);  // upto64
 // out->SetGain(((float)4)/100.0);
  mod->begin(file, out);  //out
[......]
****SD card's CS uses D3 pin

Thank you!!

earlephilhower, 2018-05-15:
Howdy,

MOD was one of those formats that everyone tweaked to make their own, so this could just be an incompatible modfile.

It's also possible that it requires >16 bit addition, which would also be incompatible AFAIK with the way the Amiga muxed channels.  You could be hearing clipping due to this.  SetOutputGain() affects things after the muxing, not before, so it would only lower the volume of clipped summed samples, if that's the case.

The other possibility is power supply/wiring related.  I've heard from several folks who got a PCM5* DAC PCB but did not hardwire several input pins.  Check #90 for more info.

Absolute file size shouldn't have any bearing on the sound output.  I picked several MODs at random from one of the large mod archives and a copy of Pump Up the Volume.mod I had lying around from decades ago (and which was much less impressive in 2018 than in 1995) when porting the Stellaplayer.  Didn't notice any problems (I think I was using the Adafruit I2S DAC+AMP at the time).

Porklaser, 2018-06-07:
I'm experiencing a similar issue with the included PlayMODFromPROGMEMToDAC example sketch. Very loud, distorted audio. (Uses the included enigma mod file)

Other example sketches play fine, including PlayAACFromPROGMEM, PlayFLACFromPROGMEMToDAC, and Play RTLLToI2SDAC.

I've made sure to check the wiring as described in the readme and #90 as described above.

SetGain() similarly reduces overall output volume, but the distortion is still there.

Using a PCM5120A breakout board connected to a generic ESP32 board.

moose4lord, 2018-09-06:
I'm also seeing this issue. I have a PCM5102A DAC connected to a NodeMCU ESP8266 as described in #90. The PlayWAVFromPROGMEM, PlayRTTTLToI2SDAC, PlayMP3FromSPIFFS and PlayFLACFromPROGMEMToDAC example sketches sound fine. The other example sketches have issues:
PlayMODFromPROGMEMToDAC -  very loud and distorted
PlayMIDIFromSPIFFS - sounds fine for a second or two, then starts to stutter and distort
PlayAACFromPROGMEM -  Homer sounds like a Dalek :)

Any idea how to debug this?

OldYin, 2019-04-11:
Been looking for a solution for this but I think it is safe to say that it has very little if anything to do with driving an external DAC. The output is just as loud and heavily distorted if you direct the output to AudioOutputI2SNoDAC. I'm guessing that it is all in the decoding of the MOD file.

earlephilhower, 2019-04-13:
The output scale was 2x too large causing massive clipping.  The latest version fixes this w/a 1 character fix (shift 15 bits, not 16 bits).  The example sounds way better now.

closed 2019-04-13 by earlephilhower