earlephilhower/ESP8266Audio #95: using ESP32 internal DAC

HumamHelfawi, 2018-05-14:
Hi,
I want to use the ESP32 internal DAC. However, I could not figure how. Using this code is giving me an error:
```
#include <Arduino.h>
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif

#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"

// VIOLA sample taken from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html
#include "viola.h"

AudioGeneratorWAV *wav;
AudioFileSourcePROGMEM *file;
AudioOutputI2S *out;

void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  delay(1000);
  Serial.printf("WAV start\n");
  file = new AudioFileSourcePROGMEM( viola, sizeof(viola) );
  out = new AudioOutputI2S(25,1);
  wav = new AudioGeneratorWAV();
  wav->begin(file, out);
}

void loop()
{
  if (wav->isRunning()) {
    if (!wav->loop()) wav->stop();
  } else {
    Serial.printf("WAV done\n");
    delay(1000);
  }
}

```

The error is:
```
WAV start
+25 0x3ffd1078
E (1175) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1017 (i2s_driver_install):i2s_num error
ERROR: Unable to install I2S drives

E (1180) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:722 (i2s_set_pin):i2s_num error
E (1191) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:995 (i2s_zero_dma_buffer):i2s_num error
E (1202) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:831 (i2s_set_sample_rates):i2s_num error
E (1214) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:831 (i2s_set_sample_rates):i2s_num error
E (1225) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1236) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1247) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1258) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1269) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1280) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1291) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1302) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1313) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1324) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1335) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1346) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1357) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1368) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1379) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1390) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1401) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1412) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1423) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1434) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error
E (1445) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./i2s.c:1263 (i2s_push_sample):i2s_num error

```

Thank you very much!

closed 2018-05-15 by earlephilhower

farkasgzoltan, 2018-10-04:
`#include <Arduino.h>
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif

#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorRTTTL.h"
#include "AudioOutputI2S.h"
//#include "AudioOutputI2SNoDAC.h"  // use sigma delta dac

const char rudolph[] PROGMEM =
"Rudolph the Red Nosed Raindeer:d=8,o=5,b=250:g,4a,g,4e,4c6,4a,2g.,g,a,g,a,4g,4c6,2b.,4p,f,4g,f,4d,4b,4a,2g.,g,a,g,a,4g,4a,2e.,4p,g,4a,a,4e,4c6,4a,2g.,g,a,g,a,4g,4c6,2b.,4p,f,4g,f,4d,4b,4a,2g.,g,a,g,a,4g,4d6,2c.6,4p,4a,4a,4c6,4a,4g,4e,2g,4d,4e,4g,4a,4b,4b,2b,4c6,4c6,4b,4a,4g,4f,2d,g,4a,g,4e,4c6,4a,2g.,g,a,g,a,4g,4c6,2b.,4p,f,4g,f,4d,4b,4a,2g.,4g,4a,4g,4a,2g,2d6,1c.6.";
// Plenty more at: http://mines.lumpylumpy.com/Electronics/Computers/Software/Cpp/MFC/RingTones.RTTTL

AudioGeneratorRTTTL *rtttl;
AudioFileSourcePROGMEM *file;
AudioOutputI2S *out;
//AudioOutputI2SNoDAC *out; // use sigmal delta on board instead of I2S DAC

void setup()
{
  WiFi.mode(WIFI_OFF);
  Serial.begin(115200);
  delay(1000);

  Serial.printf("RTTTL start\n");
  file = new AudioFileSourcePROGMEM( rudolph, strlen_P(rudolph) );
  //out = new AudioOutputI2S();
  //out = new AudioOutputI2SNoDAC(26); // I2S 0 interface, wird Sigma Delta, Pins: 22,25,26
  out = new AudioOutputI2S(0, INTERNAL_DAC);
  rtttl = new AudioGeneratorRTTTL();
  rtttl->begin(file, out);
}

void loop()
{
  if (rtttl->isRunning()) {
    if (!rtttl->loop()) rtttl->stop();
  } else {
    Serial.printf("RTTTL done\n");
    delay(1000);
  }
}
`
PlayRTTTLToI2SDAC.ino:34:31: error: 'INTERNAL_DAC' was not declared in this scope

Andr1961, 2019-03-28:
Hi,
the same mistake
error: 'INTERNAL_DAC' was not declared in this scope

XiaoKaiRen, 2019-03-29:
@Andr1961 
Add this
#define EXTERNAL_DAC 0 //default value for external DAC
#define INTERNAL_DAC 1
#define INTERNAL_PDM 2
or 

out = new AudioOutputI2S(0, 1);

Andr1961, 2019-03-29:
> @Andr1961
> Add this
> 
> `#define EXTERNAL_DAC 0 //default value for external DAC`
> `#define INTERNAL_DAC 1`
> `#define INTERNAL_PDM 2`
> or
> 
> `out = new AudioOutputI2S(0, 1);`

Hi, thanks for the hint,
I already and made,
thanks for excellent library.
Good luck.

imranaalam, 2019-10-16:
@Andr1961 on which pin are you getting output then? on which pin to connect the speaker? are you using the transistor?

petely, 2021-09-20:
I solved this for my ESP32 boards with the following modification:
  out = new AudioOutputI2S();
  **out->SetPinout(27, 26, 25); // (int bclk, int wclk, int dout)**