earlephilhower/ESP8266Audio #97: can not compile PlayMP3FromSPIFFS

Primus007, 2018-05-15:
i become this error on compiling PlayMP3FromSPIFFS:
`Arduino: 1.8.5 (Windows 7), Board: "WeMos D1 R2 & mini, 160 MHz, 4M (3M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Archiving built core (caching) in: C:\Users\Pascal\AppData\Local\Temp\arduino_cache_416573\core\core_esp8266_esp8266_d1_mini_CpuFrequency_160,FlashSize_4M3M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_46768cae23f1b87ff0730530178cdc52.a
libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::seek(int, int)':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::seek(int, int)'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:58: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::isOpen()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::isOpen()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:75: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::getSize()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::getSize()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:80: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::getPos()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::getPos()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:85: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::getFillLevel()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::getFillLevel()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:91: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::fill()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::fill()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:146: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::~AudioFileSourceBuffer()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::~AudioFileSourceBuffer()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:51: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::~AudioFileSourceBuffer()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::~AudioFileSourceBuffer()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:51: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::~AudioFileSourceBuffer()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::~AudioFileSourceBuffer()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:51: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::close()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::close()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:67: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::loop()':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:45: multiple definition of `AudioFileSourceBuffer::loop()'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:180: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::read(void*, unsigned int)':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioFileSourceBuffer.cpp:94: multiple definition of `AudioFileSourceBuffer::read(void*, unsigned int)'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:94: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, int)':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioFileSourceBuffer.cpp:26: multiple definition of `AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, int)'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:26: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, int)':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioFileSourceBuffer.cpp:26: multiple definition of `AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, int)'

sketch\AudioFileSourceBuffer.cpp.o:sketch/AudioFileSourceBuffer.cpp:26: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioStatus':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:29: multiple definition of `AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, void*, int)'

sketch\AudioFileSourceBuffer.cpp.o:F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:29: first defined here

libraries\ESP8266Audio-master\AudioFileSourceBuffer.cpp.o: In function `AudioStatus':

F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:29: multiple definition of `AudioFileSourceBuffer::AudioFileSourceBuffer(AudioFileSource*, void*, int)'

sketch\AudioFileSourceBuffer.cpp.o:F:\Eigene Dateien\Arduino\libraries\ESP8266Audio-master\src/AudioStatus.h:29: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.
`
i am using a wemos d1 mini.  PlayRTTTLToI2SDAC works.  what can i do?

earlephilhower, 2018-05-15:
Did you install this under ~/Arduino/libraries?  It looks like you've got multiple copies somehow in your build chain.

Primus007, 2018-05-16:
now it works. i think this was problems from Win7.

closed 2018-05-16 by Primus007

spspider, 2019-07-06:
i have same problem. not compiling at all (examples)
i use 2.5.0 version of  hardware, and latest version of ESPAudio i have just only exit status 1 with some internal errors, but it seems that doesnt influence on it. compilation stops on ESP8266_Spiram