earlephilhower/ESP8266Audio #100: How to connect the esp32?

Primus007, 2018-05-23:
i have successfully wired my pcm5102 to a wemos d1. 
Now i want to use a ESP32. I have connected:
BCK Gpio26
LCK Gpio25
DIN Gpio22
Power:
3,3v + to ESP Vin and DAC VCC
3,3v - to ESP GND and DAC GND

But i hear no sound.
please can  help anyone?

RemyEE, 2018-06-22:
Using PlayMP3FromSPIFFS example, I changed AudioOutputI2SNoDAC to AudioOutputI2S (include header file, class file name), the wiring just same as your
BCK Gpio26
LCK Gpio25
DIN Gpio22
It works for me on ESP32.
I used Adafruit UDA1334, but it should works for another I2S DAC module.
Good lucks.

aviosipov, 2019-02-23:
I addition, SCK have to be connected to GND

tfry-git, 2019-04-16:
For what it's worth, if the issue is that you get more noise than sound, it could be the bit order. To get my PT8211 to work, I had to change ```I2S_COMM_FORMAT_I2S_MSB``` to ```I2S_COMM_FORMAT_I2S_LSB```, here: https://github.com/earlephilhower/ESP8266Audio/blob/d0089c1f4acbcca48ef852400dfdcfd75ff06c8c/src/AudioOutputI2S.cpp#L56 .

zircothc, 2019-05-09:
> For what it's worth, if the issue is that you get more noise than sound, it could be the bit order. To get my PT8211 to work, I had to change `I2S_COMM_FORMAT_I2S_MSB` to `I2S_COMM_FORMAT_I2S_LSB`, [...]

Wow, tfry-git you save me hours.
 I was getting mad testing several pt8211 and a pair of esp32 and noise was horrible.
Thank you

FrankHovis, 2021-04-03:
How do I make this change on an esp8266?