earlephilhower/ESP8266Audio #101: Using Amplifier and TTL converter at same time

abdollahpour, 2018-06-25:
I use TTL/USB convert and Amplifier (PAM8302) at the same time. It means the RxD is connected to TTL converter and the Amplifier. When I disconnect the TTL connection, the sound volume is amazingly louder and when I add a resistor to limit the current for TTL converter, it stops working. Any idea to fix the problem without adding noise to the audio? Can I add a transistor switch to for TTL converter to trig it with a little bit current?
Thanks for your great lib and perfect support.

mikaelsundin, 2018-07-04:
You can use PAM8302 shutdown pin(SD) to one GPIO with a resistor pull-down.
Then you must active draw pin high to enable PAM8302, something the bootloader wont normally do.

GPIO15 for example, must be pulled low anyway to enter uart bootloader or boot from flash.

abdollahpour, 2018-07-05:
Thanks for the solution. I exactly do the same thing. But my problem is something else. I can disable PAM8302 when I use the TTL converter (CP2104), but I can't disable TTL converter (RxD pin) when the amplifier is active. When I cut off the wire to CP2104, the sound is much louder.

earlephilhower, 2018-07-05:
The I2S pins are fixed on the ESP8266, not like the ESP32 where you can move them around a bit.

That said, if you are using an I2S DAC then it's a binary, digital signal.  Either it makes it into the DAC or not.  Loads on the I2S-DOUT pin don't matter as long as the swing from 0->1->0 meets all the Vin(low,high) thresholds.

A volume change would indicate that your V(analog) to the DAC/AMP is too low, probably due to the USB -> TTL converter using enough current to drop the voltage presented.  If you can, and the amp supports it, using the raw 5V from the USB port, not any regulated 3.3V, for the analog V supply may help.  OTW you need a separate power source.

mikaelsundin, 2018-07-06:
You could connect a tri state output logic buffer/mux between cp2104 and control it with same pin as Pam shutdown.

mikaelsundin, 2018-07-06:
I think a n-mosfet is enough
Mos Source - esp8266 rx + audio out
Mos Drain - cp2104
Gate - 3.3v

mikaelsundin, 2018-07-06:
I think a n-mosfet is enough
Mos Source - esp8266 rx + audio out
Mos Drain - cp2104
Gate - GPIO2 for example. pulled high under programming and low when running audio.

Also a pull-up is needed at esp8266 rx - somewhere around 10k should be enought.
This is the same trick as running I2C bus with different voltage levels..

abdollahpour, 2018-07-07:
Thanks, you guys are great. I have no ideal about tri state buffer, it’s difficult for me, but I definitely test the Mosfet one. I have to mention the sound level is not bad already but because my device is for outdoor, even a little bit louder is important.

abdollahpour, 2018-07-16:
The n-mosfet solution works beautifully. Just one more thing, is there any complimentary cheap P-Channel for 2N7002. Because then I can switch PAM8302 (SD pin) on when TTL converter is off and vice versa using just one pin. Does this solution work? Thanks

mikaelsundin, 2018-07-25:
I am not aware of a generic 2N7002 p-mos complimentary. But it exist many p-mos in SOT23 package.
You can use 2N7002 + a pull-up to create a inverting gate.

Mos Source - GND
Mos Drain - output, connectet the pull-up between this pin and vcc.
Gate - input signal.

It can also be good to have a pull-down on gate to discharge the gate capacitance if the input is free floating.

closed 2020-08-08 by earlephilhower