earlephilhower/ESP8266Audio #102: [Feature Request] Read .Wav Files from SD Card

nathanRamaNoodles, 2018-06-29:
Thank you so much for this amazing library. :)
I was interested in your PlayWavFromPROGMEM example, and I am currently on a project where I need more space.  
Is it possible to put the char array in the SD card. Maybe you could adjust **AudioGeneratorWAV.cpp** for ASCII characters?  This would definitely be cool to run because .wav files don't require an I2S chip., instead, they just use your n3904 transistor method.  

Thx again.

earlephilhower, 2018-07-02:
There's already an `AudioFileSourceSD` object.  Just copy your file to SD using the FAT filesystem, use the `AudioFileSourceSD` object (pass it the filename on the SD card in constructor), and it can then feed into any of the playback decoders (WAV, MP3, etc.).

closed 2018-07-02 by earlephilhower

imranaalam, 2019-10-16:
@nathanRamaNoodles i am facing the same problem. please share your code. on what pins are you getting the output?