earlephilhower/ESP8266Audio #103: Update or delete Platformio library entry

Oliv4945, 2018-07-10:
Hi,

First, thank your for your library: it works like a charm when I2S clock & data pin are not inverted on my side :)

[Platformio ](https://platformio.org/lib/show/1964/ESP8266Audio/examples) library is a outdated, which can lead people to think that the library is not working because you corrected lot of bugs since November, 2017.
It would avoid confusion if you update or delete it :)

earlephilhower, 2018-07-10:
Howdy,

I didn't create that Platform.IO library, so I'll have to figure out who did and how it would be updated.  Have you any libraries there?  Would you be able to give any pointers?

From a quick glance I need to bump the library version in the repo, PIO doesn't seem to grab updates to an existing rev.

Oliv4945, 2018-07-11:
I never published any library but I will check it for you :)

earlephilhower, 2018-07-11:
Thanks, @Oliv4945 , but by bumping the rev number I was able to get the latest version "1.1" up on Platform.IO.

Looks like we're good now.  Thanks again for letting me in on this, I'd never have known otherwise.

closed 2018-07-11 by earlephilhower

Oliv4945, 2018-07-12:
Yes I so that yesterday, thank you !