earlephilhower/ESP8266Audio #104: Stack smashing protect failure! on ESP32 using internal dac

zinefer, 2018-07-11:
I get a ```Stack smashing protect failure!``` and then the esp32 reboots over and over. However, it seems like after 30ish reboots it will bypass this issue and play my mp3 without issue until I reboot the esp32 and then it gets stuck in the reboot loop again.

Here is a decoded stack exception:
```
Decoding stack results
0x4008db10: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 139
0x4008dceb: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 148
0x400e32f4: __stack_chk_fail at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./stack_check.c line 36
0x400d30f7: AudioFileSourceID3::read(void*, unsigned int) (C:\Program Files at x86)\Arduino\libraries\ESP8266Audio\src\AudioFileSourceID3.cpp line 240
0x400d36ac: AudioGeneratorMP3::Input() (C:\Program Files at x86)\Arduino\libraries\ESP8266Audio\src\AudioGeneratorMP3.cpp line 122
0x400d380b: AudioGeneratorMP3::loop() (C:\Program Files at x86)\Arduino\libraries\ESP8266Audio\src\AudioGeneratorMP3.cpp line 192
0x400d1ba9: loop() at C:\Users\James\Documents\soundtest/soundtest.ino line 65
0x401221f4: loopTask(void*) at C:\Users\James\Documents\Arduino\hardware\expressif\esp32\cores\esp32\main.cpp line 18
```

earlephilhower, 2018-07-11:
Looks like it's related to a specific MP3.  Can you share it?  My guess is my ID3 decoder is barfing on some format it's never encountered before.

zinefer, 2018-07-11:
Sure!

[Tetris-GameBoy-TypeA.zip](https://github.com/earlephilhower/ESP8266Audio/files/2186081/Tetris-GameBoy-TypeA.zip)

earlephilhower, 2018-07-11:
Simple to reproduce on my host tests.  Was a 1-byte write out of bounds when >64 bytes in a single ID3 field.  Latest commit fixes this and runs the attached MP3 fine.

closed 2018-07-11 by earlephilhower

zinefer, 2018-07-12:
Thanks a bunch! Everything is working well on my end now.

commit 7d88ebca (2018-07-11): Fix buffer overflow in ID3 decode

When the ID3 tag had >64 bytes of data for an element, there was a
1-byte out of bounds access, causing a crash at some later point in
time.  Properly limit writes to the ID3 field buffer.

Fixes #104
commit c0ec2616 (2018-07-11): Fix buffer overflow in ID3 decode

When the ID3 tag had >64 bytes of data for an element, there was a
1-byte out of bounds access, causing a crash at some later point in
time.  Properly limit writes to the ID3 field buffer.

Fixes #104