earlephilhower/ESP8266Audio #105: playing at twice the speed since platformio update

h3ndrik, 2018-07-12:
tested: modified version of examples/StreamMP3FromHTTP.ino
with an MAX98357A  I2S-board

compiled with platformio 

> platform = espressif32@1.0.2

 it works as intended, same code with updated platform: sounds like double the sample rate

changing AudioOutputI2S::SetRate to:
    i2s_set_clk((i2s_port_t)portNo, AdjustI2SRate(hz), I2S_BITS_PER_SAMPLE_16BIT, (i2s_channel_t) 1);
works (note the change from 2 channels to 1). as does chaning channel_format to I2S_CHANNEL_FMT_ONLY_RIGHT

is this a bug? what am i doing wrong?

flying4fun, 2018-07-20:
I wonder if this is what I'm running into.  Although making that change in AudioOutputI2S::SetRate didn't work for me.  I still get a huge amount of distortion and sounds like it's playing way too fast.

earlephilhower, 2018-07-20:
Just to confirm, when things were updated was it both the ESP SDK as well as the audio library, or only the audio library?

Most likely this has something to do with the I2S port configuration which just calls the SDK to set things like # of bits, # of channels, and speed.

If I understand things, the same sample on ESP8266 runs fine but on the ESP32 it's 2x as fast, yes?  That vastly reduces the possible area it could be in (i.e. only `#ifdef ESP32` blocks)...

flying4fun, 2018-07-20:
For me, yes I updated the ESP SDK as well as this audio library.  

Yes, ESP8266 works as it should.  And it appears to be playing twice the speed ONLY on the ESP32.

h3ndrik, 2018-08-02:
the problem seems to be i2s_push_sample() in AudioOutputI2S::ConsumeSample(...):

the function returns 2 for me, but a sample should be 4 bytes long (2 channels, 16bit each).
previously (with mentioned platformio platform 1.0.2) it returned 4!

as i2s_push_sample is deprecated anyways, i'm trying to replace it with something like:
    size_t bytes_written = i2s_write((i2s_port_t)portNo, &s32, 4, &bytes_written, portMAX_DELAY);
    return bytes_written;

flying4fun, 2018-08-03:
h3ndrik, thanks!  That does indeed fix it.

earlephilhower, 2018-08-03:
Thanks again for the debug and fix!

closed 2018-08-03 by earlephilhower