earlephilhower/ESP8266Audio #109: WebRadio example - not working web generator

jgrulich, 2018-07-20:
I'm using Windows Arduino 1.8.5 and 2.4.1 with IwIP Variant v2 Higher Bandwitch and like to test the WebRadio sample and found, that the web generator doesn't work as expected. The web server is active, the web socket port 80 is responding and it does the work in the part of program, but the web page doesn't appear.
In the serial console is as follows:
> 
Running for 159 seconds ...Free mem=7152
Running for 160 seconds ...Free mem=7152
Sending INDEX...Free mem=6336
Sent INDEX...Free mem=1536
Running for 161 seconds ...Free mem=6960
Running for 162 seconds ...Free mem=6960
> 
Any idea, what to test? I've tried several web browsers and tested mobile Android browser, but not lucky.
I've tested even another ESP8266 core version, but 2.3.0 fails to compile.

earlephilhower, 2018-07-20:
You're running out of memory.  Please try the GIT HEAD version of the esp8266 Arduino core which includes changes which give about 4K extra heap and which also has several core memory leak fixes.

jgrulich, 2018-07-22:
Yes, it's a lack of memory. With 2.4.0 core it sometimes works just after the reset and once, or twice the page appears. But than it hangs, or restarts. The 2.4.0. core has wifi signal issues and therefor can't be used for my purpose.
I've tested the git version of ours, but it is not as memory efficient as 2.4.0 core and it work only sometimes. And even more there is issue with the building, this core builds every time from scratch and it takes a quite long time.

moose4lord, 2018-09-21:
I had some problems with the Web Radio example too. The web server part of the code does not send Content-length in the response header, which can cause the browser to hang or display a blank page. I've attached a hacked version of the sketch that fixes this issue. You can give that a try.

Also, I had all kinds of problems with the web radio stations I tried to connect to. Some worked pretty well, but many just didn't work well at all. I ended up using the VPR (Vermont Public Radio) stream to do my testing. They have pretty stable streams at a few different bit rates to play with. They offer three mp3 streams (at bit rates of 24kbps, 64kbps and 96kbps) at these URLs:
http://vpr.streamguys1.com/vpr24.mp3
http://vpr.streamguys1.com/vpr64.mp3
http://vpr.streamguys1.com/vpr96.mp3
The 24kbps and 96kbps streams work pretty well (although both stutter occasionally), but the 64kbps stream would cause my ESP8266 to reboot after a few seconds. Weird.

Overall I found the web radio application to be pretty fragile.

[WebRadio.zip](https://github.com/earlephilhower/ESP8266Audio/files/2405849/WebRadio.zip)

jgrulich, 2018-09-22:
Hi @moose4lord , good point to the web issue. I've tested and It works now with EPS8266 and latest git core, but still not with ESP32 and latest git core.

 You're right that ESP as radio, or streaming app is very unstable and hardly be used.

 ESP32 somehow works, because there is much more RAM for buffer, but still with a lot of gaps and reboots. Don't know why, but the buffer may be only 63kB, even there is over 150kB free heap. When used over 63kB, it reports the error with the space allocation at startup.

 The best performance now I have is the ESP8266 with the SPIRAM. But it still need to be patched with a lot of modifications to reconnect the stream when the connection is lost until it plays the stream from the buffer and refill the buffer again.

earlephilhower, 2018-09-23:
@jgrulich The buffer is indexed via a uint16_t, so it can't point to more than 64K.  64K is 20K more than the total heap in the 8266 so it wasn't forseen. :(  I'm happy to accept a patch moving it to a 32-bit, though. :)

jgrulich, 2018-09-26:
Guessed that it was already solved here (#117), but it doesn't work in this WebRadio example. In the  	StreamMP3FromHTTP example the buffer may be up to around 110K. This seems to be enough for smooth playback even at noisy and unstable WiFi.

closed 2020-11-18 by jgrulich