earlephilhower/ESP8266Audio #110: I2S rate too fast on ESP32?

flying4fun, 2018-07-20:
Tried several examples.  I get chipmunks using my ESP32.  ESP8266 is just fine.  Latest example is PlayMODFromPROGMEMTODAC

I'm using the latest ESP32 libraries and ESP8266Audio as of today.

Any ideas on what might be happening?

flying4fun, 2018-08-03:
Applied this change to my copy of the library and it does resolve the issue.  Thanks!

closed 2018-08-03 by flying4fun

Mael19, 2018-08-07:
Hello, I’m having the same issue (I think). When streaming mp3 with the esp32 the sound goes “too fast” and there is many many errors/bugs.” Updating the library doesn’t seems to change anything... do you have an idea ?

earlephilhower, 2018-08-07:
I think you have an older version of the sound libs.  The double speed was because Espressif made an incompatible change to their libraries in thirty latest release.  @h3ndrik submitted a fix which was committed, fixing it by using a different call.

Mael19, 2018-08-08:
I don’t quite understand... I updated the library just yesterday but nothing changed... I double checked the corresponding line in the .cpp too !

Mael19, 2018-08-18:
I'm sorry to bother again but I still haven't been able to solve my problem... Can someone help me ?