earlephilhower/ESP8266Audio #111: Using I2S and CS4344

abdollahpour, 2018-07-21:
I'm trying to use the cheap digital-to-analog amplifier, CS4344. Unfortunately, I can not hear anything. I've tried any possible combination. Could you please help me?
Pins are:

   -- SDIN
   -- DEM
   -- LRCLK
   -- MCLK

The other pins are defined in the datasheet:
http://www.mouser.com/ds/2/76/CS4344-45-48_F2-472818.pdf

jgrulich, 2018-07-22:
This decoder is not suitable for this library, because it needs the MCLK to be sourced externally from the master, because it need to be synchronous with the SDIN and LRCLK. The DEM/SCLK pin is not necessary for operation and may be grounded. But this lib uses asynchronous data stream with MCLK generated internally in the DAC like in the PCM510x.

earlephilhower, 2018-07-22:
Yup. @jgrulich is right from what I saw on the datasheet.  Unlike most chips which have an internal PLL to generate the proper internal hi-speed clocking, this one needs to be driven from outside.  The ESP8266 definitely can't do this, maybe the ESP32 might be able to.  You'd need an external clock generator on the 8266 and to change the Arduino core libs to use external I2S clocking, which would definitely negate any cost savings on the DAC chip itself...

closed 2018-07-22 by earlephilhower

jgrulich, 2018-07-24:
Not really sure with the CS4344 operation, but PCM5102 may work with the external clock too.
You may try to use some low cost clock generator with the following frequencies to the MCLK pin:
11.2896, 16.9344, 22.5792, 33.8680, 45.1580 MHz
Than connect the pins SDIN, SCLK and LRCK as usual.

abdollahpour, 2018-07-24:
To be honest, I'm a software guy, I'm not very familiar with Electronic, I have no idea about the clock generator. Also, In my case, PCB is very small, I did not even had enough space for ESP-12, and I use ESP8266ex itself (opened the module), plus the other components. I don't think I can add another IC to my board.

jgrulich, 2018-07-24:
The clock generator is like this:
https://cz.mouser.com/ProductDetail/Kyocera-Electronic-Components/KC2016K112896C10E00?qs=sGAEpiMZZMtldj7qu1ydrWqE5pfwGz%252b46pL2R7mMj8lQ4CgCcrjzAQ%3d%3d
and it's very small, only 2.0x1.6mm.

abdollahpour, 2018-07-24:
Oh, if it's Crystal Oscillator I already have two on my PCB for ESP & TTL converter! I think I have enough space for the third one! And they are quite cheap, just 0.05$! Which one of those frequencies I have to choose with what capacitor (normally it's with two capacitor around)? Does this frequency related to Bitrate of the WAV file? How much the bitrate has to be? Thanks

earlephilhower, 2018-07-24:
Crystal oscillators need to be loaded properly and excited to start up oscillation.  They're also not TTL or CMOS devices, so you'll need buffering to get a full-swing signal out of it.  Once you get that working, which is often a pain in the rear since it's analog and high-frequency, you need to rewrite the Arduino core libs to allow I2SCLK to be an input and to drive the internal I2S circuitry off of the pin.  You'll need, then, to ensure that the MCLK is one of the supported multiples of the LRCLK for the output sampling frequency (and if you try and run a sample with a rate that's not supported, too bad), and that the edges all line up properly WRT rising/falling of mclk.

If you're building 1,000, it may be worth the work.  If you're building 1 or 100, unless your time has no value I'd suggest finding an I2S DAC with internal clock generation.

jgrulich, 2018-07-24:
Please don't mix crystal, which has two pins and need some stuff around, and crystal oscillator, which is in link above and has usually four pins (GND, Vcc, EN and OUT) and doesn't need anything around and may be directly connected to the CS4344 MCLK input. I have no experience with CS4344, but when used PCM5102A, it works without any change of the ESP code. It's recommended to have the MCLK to be synchronous with LRCLK and SCLK, but it works not sync. too. The SCLK is used to sync. data input and usually it's enough, when the crystal oscillator has low jitter.
About the frequency, look in the table of CS4344 datasheet to look which is the right one for the used sampling frequency. As I've found in the I2S lib, there is only one common sampling frequency used. The setup is:
"
  bps = 16;
  channels = 2;
  SetRate(44100); // Default
"
Than you may use any of the frequencies mentioned above, but two of them are most common:
11.2896 and 22.5792 MHz.

But as @earlephilhower mentioned, for small quantities is better to use PCM5101A, which has same THD and noise performance as your CS4344 and has output at common ground ( there is no DC bias ) and doesn't need large coupling caps, which simplifies the design and reduces the PCB space. BTW, PCM5101A is even cheaper than CS4344 at Mouser.

earlephilhower, 2018-07-24:
Mybad, @jgrulich!  Been too long since I had to do PCB design, and xtals (not osc's) were always used for cost and size so I didn't catch the difference.  This was through-hole design, so that gives an idea how long ago that was....

However I would bet that @abdollahpour only has a simple crystal on board driving the ESP8266.  All the schematics I've seen use the pins 27 and 28 labelled xtal_in/_out and a (26?)MHz xtal..

jgrulich, 2018-07-24:
@earlephilhower , that's fine, seems that we're the same age.
Yes, you're right. ESP uses simple crystal, not oscillator. But I've offered to use the oscillator instead because of simplicity of use. You don't need to care about the xtal load, wave squaring, etc. And it's as small as xtal itself, which saves another board space and @abdollahpour has a lack of board space as seems to me.