earlephilhower/ESP8266Audio #113: How to play mp3 file from a website

mohamad-jawad, 2018-07-25:
I have tried to use the example "StreamMP3FromHTTP" with the following link "http://streaming.shoutcast.com/80sPlanet?lang=en-US" and it worked successfully.

but what i want is to play an mp3 sound from a website (non streaming) what i can do ?

earlephilhower, 2018-08-01:
Just replace the `AudioFileSourceICYStream` with `AudioFileSourceHTTPStream` and get rid of the metadata callback and it should just work.

closed 2018-08-01 by earlephilhower

mohamad-jawad, 2018-08-02:
I replaced each `AudioFileSourceICYStream` with `AudioFileSourceHTTPStream` and commented out all the lines in MDCallback.
I am not getting any error but cant get any sound out of my esp32.
This is the output of my serial : 

> ...Connecting to WiFi
> ...Connecting to WiFi
> ...Connecting to WiFi
> Connected
> +0 0x3ffd1308
> Running for 12417 ms...
> STATUS(buffer) '2' = 'Refilling buffer'
> STATUS(buffer) '3' = 'Buffer underflow'
> MP3 stop, len==0
> MP3 done
> MP3 done
> MP3 done

This is what my code looks like
```
#include <Arduino.h>

#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif
#include "AudioFileSourceHTTPStream.h"
#include "AudioFileSourceBuffer.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"

// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.

// Enter your WiFi setup here:
const char *SSID = "....";
const char *PASSWORD = "....";

// Randomly picked URL
const char *URL="https://ccrma.stanford.edu/~jos/mp3/cello.mp3";

AudioGeneratorMP3 *mp3;
AudioFileSourceHTTPStream *file;
AudioFileSourceBuffer *buff;
AudioOutputI2SNoDAC *out;

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{/*
  const char *ptr = reinterpret_cast<const char *>(cbData);
  (void) isUnicode; // Punt this ball for now
  // Note that the type and string may be in PROGMEM, so copy them to RAM for printf
  char s1[32], s2[64];
  strncpy_P(s1, type, sizeof(s1));
  s1[sizeof(s1)-1]=0;
  strncpy_P(s2, string, sizeof(s2));
  s2[sizeof(s2)-1]=0;
  Serial.printf("METADATA(%s) '%s' = '%s'\n", ptr, s1, s2);
  Serial.flush();*/
}

// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void StatusCallback(void *cbData, int code, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  // Note that the string may be in PROGMEM, so copy it to RAM for printf
  char s1[64];
  strncpy_P(s1, string, sizeof(s1));
  s1[sizeof(s1)-1]=0;
  Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
  Serial.flush();
}

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Connecting to WiFi");

  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(SSID, PASSWORD);

  // Try forever
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("...Connecting to WiFi");
    delay(1000);
  }
  Serial.println("Connected");

  file = new AudioFileSourceHTTPStream(URL);
  file->RegisterMetadataCB(MDCallback, (void*)"ICY");
  buff = new AudioFileSourceBuffer(file, 2048*16);
  buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
  out = new AudioOutputI2SNoDAC();
  mp3 = new AudioGeneratorMP3();
  mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
  mp3->begin(buff, out);
}

void loop()
{
  static int lastms = 0;

  if (mp3->isRunning()) {
    if (millis()-lastms > 5000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
     }
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}

```

kelliott121, 2020-11-03:
I can confirm that I see the same issue attempting to play audio from a hosted MP3 file.

simpleverso, 2022-11-27:
Same issue