earlephilhower/ESP8266Audio #114: 16kHz 8bit WAV not working

positron96, 2018-07-29:
Hello. I created a set of WAV samples of the same file but different bitrates and frequences:
 * 16 bit signed le  16khz 
* 16 bit signed le  22khz 
* 16 bit signed le  44khz 
* 8 bit unsinged  16khz 
* 8 bit unsigned  22khz
* 8 bit unsigned  44khz 

All of them work except 8bit 16 khz. Here is the file archived (github did not accept raw WAV).
[test_8u_16.zip](https://github.com/earlephilhower/ESP8266Audio/files/2239019/test_8u_16.zip) . It's a horn for model train.

All other files are generated from it with ffmpeg similar to this: ``` ffmpeg -i test_8u_16.wav -acodec pcm_s16le -ar 44k test_16sl_44.wav```

Is there any particular reason for it not to work? Also, can I get any debug output. e.g. error description to Serial?

earlephilhower, 2018-07-29:
What are you using to play back these samples?  The SW sigma-delta, or a real I2S DAC (which one)?  Or are you on a ESP-32 w/internal DAC?

What doesn't make sense is that 8 bit is *always* promoted to 16-bit on the 8266.  So 8bit @16khz == 16b @ 16khz.  And other than setting the frequency, 16khz is no different than 44khz as it just sends samples.

Now there are some DACs which do not support certain frequencies.  The datasheet should list the supported frequencies.

I don't really have any debug options as there's really no debugging on the I2S interface (it just sends whatever its told, there is no feedback on the interface wires at all).

positron96, 2018-07-30:
Hello and thank you for response. I am playing them on I2SNoDAC option (with a 2n3904 transistor) on a ESP8266 (Wemos D1 Mini). 

What I made was a quick and dirty modification of PlayMP3FromSPIFFS example to play every file from SPIFFS. Here is the code:

```
#include <Arduino.h>
#ifdef ESP32
  #include <WiFi.h>
  #include "SPIFFS.h"
#else
  #include <ESP8266WiFi.h>
#endif

#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2SNoDAC.h"

// To run, set your ESP8266 build to 160MHz, and include a SPIFFS of 512KB or greater.
// Use the "Tools->ESP8266/ESP32 Sketch Data Upload" menu to write the MP3 to SPIFFS
// Then upload the sketch normally.  

// pno_cs from https://ccrma.stanford.edu/~jos/pasp/Sound_Examples.html

AudioGeneratorWAV *wav;
AudioFileSourceSPIFFS *file;
AudioOutputI2SNoDAC *out;

int nfiles = 0;
int cfile = 0;
String files[9];

void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  delay(1000);

  SPIFFS.begin();
  Dir dir = SPIFFS.openDir("");  
  while (dir.next()) {
      Serial.print(dir.fileName());
      files[nfiles] = dir.fileName();
      nfiles++;
  }

  Serial.printf("Sample WAV playback begins...\n");
  
  file = new AudioFileSourceSPIFFS( files[cfile].c_str() );
  Serial.printf("Playing %s \n", files[cfile].c_str() );
  out = new AudioOutputI2SNoDAC();
  wav = new AudioGeneratorWAV();
  wav->begin(file, out);
}

void loop()
{
  if (wav->isRunning()) {
    if (!wav->loop()) { wav->stop(); delay(1000);  }
  } else {
    Serial.printf("WAV done\n");
    delay(2000);
    delete wav;
    delete file;
    cfile = (cfile+1) % nfiles;
    file = new AudioFileSourceSPIFFS( files[cfile].c_str() );
    Serial.printf("Playing %s \n", files[cfile].c_str() );
    wav = new AudioGeneratorWAV();
    wav->begin(file, out);
  }
}
```

positron96, 2018-08-02:
Hm, it appears that the library has problems with Adobe Audition-encoded files. After I re-encoded the file to the same depth/bitrate with ffmpeg it played well.

earlephilhower, 2018-08-03:
This is most likely a problem in the WAV file decoder on my end.  I see the .WAV file has an XML encoded section(!) which I've never encountered before.  I'll update the bug title and see what I can see looking at the Adobe WAV and my source.

closed 2018-08-03 by earlephilhower

positron96, 2018-08-06:
Awesome, thank you!

commit 955b1d29 (2018-08-03): Fix WAV header decode, add WAV host test

Fixes #114

Adobe Audition WAV files have a slightly different but legal header
format.  Enable that format (only for PCM, not a/ulaw or anything else)
to be decoedd properly.

Add in a WAV object test that can run on hosts and valgrind.