earlephilhower/ESP8266Audio #119: Enhancement request: AudioFileSourceA2DP + Demo sketch

tueddy, 2018-08-20:
Hi Earle,

ESP8266Audio works like a charme on ESP32! 
One more thing is missing: A source for classic Bluetooth audio.
ESP32 has everything builtin for receiving/playing audio via bluetooth A2DP. There is a sample in IDF but i never got this working in Arduino IDE before. Now i have a fully working demo playing audio via bluetooth speaker connection, tested with iOS, MacOS and Android. I've attached an example sketch for latest arduino-ESP revision.

It would be very helpful to see this integrated into ESP8266Audio, so i can choose different sources and outputs and mixing them. Unfortunatly it's out my capabilities to integrate that and make a pull request.
[image]
[image]

The example sketch is running on ESP32, using a hardware I2S DAC, standard PINs 25,26, 27 + CS on Pin12.

Hope this is starting point for an AudioFileSourceA2DP.. Let me know what you think or if you need more information..
[a2dp_sink.zip](https://github.com/earlephilhower/ESP8266Audio/files/2303692/a2dp_sink.zip)

Best regards
Dirk

Martin-Laclaustra, 2020-11-01:
Additional code that could help to implement this is at:
https://github.com/pschatzmann/ESP32-A2DP
Please see:
https://github.com/pschatzmann/ESP32-A2DP/issues/11

Regards

harryberlin, 2024-05-29:
is a2dp implemeted yet?
exist an a2dp receiver example?