earlephilhower/ESP8266Audio #123: Clicks at the beginning and end of sample playback

moose4lord, 2018-09-01:
Is anyone experiencing a clicking or popping sound at the beginning and end of sample playback? Even with the pretty straightforward PlayWAVFromPROGMEM example sketch I get a distinct clicking at the beginning and end of the viola playback. The sample plays fine, but the clicks are annoying.

I'm using the two-capacitor-and-a-resistor NoDAC low pass filter described here:
http://blog.dspsynth.eu/audio-hacking-on-the-esp8266/
driving a small powered speaker, which seems to work ok. Would Earl's 1-T output driver fix my problem? Or a real DAC? Or is this something inherent in the I2S hardware/software?

Thanks

moose4lord, 2018-09-05:
I'll answer my own question. :) I bought a PCM5102A DAC on eBay and after connecting it as described in #90 the clicking at the beginning and end of playback disappeared. Must have had something to do with the two-capacitor-and-a-resistor NoDAC output filter I was using.

jgrulich, 2020-11-18:
Hi @earlephilhower , is there any way how to keep the decoder running even when there is no data to keep the "zero" DC level?
The pops are caused by starting / stopping of decoder which cause the output PWM data to go from roughly 50% to 0% duty and that results in DC offset to go from 1.5V to 0V.
I've tried to find the code which controls the decoder to keep him running all the time the decoder object exists, but no lucky.
Please put me the right direction.