earlephilhower/ESP8266Audio #127: Odd issue with WAV files using 22050 sample rate

moose4lord, 2018-09-11:
Hello, I have been running some ESP8266Audio tests with an ESP8266 and PCM5102 DAC, and have been struggling with an odd issue that cropped up when I tried to use WAV files recorded at 22050 samples/second. Most of the time the WAV plays fine, but sometimes it's garbled. I have a test sketch which loops a WAV file and when I use a 22050 samp/sec WAV file, seemingly randomly, about 20% of the time, the sound is mangled. The same audio sample recorded at 44100 samp/sec works fine.

I've attached my test sketch and a couple WAV files (the same sound recorded at 44100 samp/sec and again at 22050 samp/sec). I'm hoping someone with a similar setup can run the sketch and see if they see the same thing. It'd be nice to know if it's some kind of weird race condition peculiar to my hardware setup, or a bug with the software.

Thanks
[ESP8266Audio_test.zip](https://github.com/earlephilhower/ESP8266Audio/files/2372854/ESP8266Audio_test.zip)

earlephilhower, 2018-09-11:
Looking at the datasheet, I don't think it officially shirts that sample rate.  Many DACs have PLLs that are tuned to specific frequencies (and the digital filters to make them sound good).  This one, I don't see 22k as supported, so you might end up with it not getting the clocking right...

moose4lord, 2018-09-12:
Thanks, Earle. Good catch on the supported sample rates in the datasheet. 22050 samp/sec is a pretty common rate, so I just assumed it was supported.

Out of curiosity I created a few WAV files at different sample rates. This is what I found:
16000 samp/sec - occasionally mangled audio
22050 samp/sec - occasionally mangled audio
32000 samp/sec - occasionally mangled audio
44100 samp/sec - always sounds good, rock solid
48000 samp/sec - occasionally mangled audio

The 16k and 22k WAVs aren't supported (for 16-bit samples) according to the datasheet, so I accept they're flakey. But the 32k and 48k WAVs are supported, but still intermittently emit garbled audio. Weird. 44.1k WAV files are the only ones I get good, repeatable results with.

Anyway, just FYI. Thank you for your help.

closed 2020-05-11 by earlephilhower