earlephilhower/ESP8266Audio #128: Slow Audio on ESP32 internal DAC using Mixer output

darookee, 2018-09-12:
For a project I need to be able to play music and occasionally play sound effects without stopping the music. The mixer output seems to be doing that. Unfortunately on the ESP32s internalt DAC it causes the music to play very slow.

I used the following code to test if it realy is the notifications/effect playing and not the SD card speed or the WiFi or other input checking I do in my original program.

```
// just to satisfy platformio dependencies
#include <HTTPClient.h>

#include <SD.h>
#include <SPIFFS.h>

// audio
#include <AudioGeneratorMP3.h>
#include <AudioOutputI2S.h>
#include <AudioOutputMixer.h>
#include <AudioFileSourceSD.h>
#include <AudioFileSourceSPIFFS.h>

AudioGeneratorMP3 *music;
AudioGeneratorMP3 *notification;
AudioFileSourceSD *sdFile;
AudioFileSourceSPIFFS *notiFile;
AudioOutputI2S *out;
AudioOutputMixer *mixer;
AudioOutputMixerStub *musicOut;
AudioOutputMixerStub *notifOut;

unsigned long waitBeforeNotification = 2500;
uint8_t doNotify = 1;

void setup() {
    Serial.begin(115200);

    SD.begin();
    SPIFFS.begin();

    out = new AudioOutputI2S(0, 1);
    mixer = new AudioOutputMixer(32, out);

    sdFile = new AudioFileSourceSD("/01ab3c35a918aae4ac0669bb1c243c07.mp3");
    music = new AudioGeneratorMP3();
    musicOut = mixer->NewInput();

    music->begin(sdFile, musicOut);
    Serial.printf("Starting playback...\r\n");
}

void loop() {
    if (music && music->isRunning()) {
        if (!music->loop()) {
            music->stop();
            delete music;
            music = NULL;
        }
    }

    if (notification && notification->isRunning()) {
        if (!notification->loop()) {
            notification->stop();
            delete notification;
            notification = NULL;
        }
    }

    if (1 == doNotify && !notification && millis() >= waitBeforeNotification) {
        if (SPIFFS.exists("/ready.mp3")) {
            notiFile = new AudioFileSourceSPIFFS("/ready.mp3");
            notification = new AudioGeneratorMP3();
            notifOut = mixer->NewInput();

            notification->begin(notiFile, notifOut);
            Serial.printf("Notifying...\r\n");
        }
    }
}
```

If `doNotify` is set to `1` the music plays slow, if it is `0` the music plays at normal speed. My platformio configuration looks like this: 

```
[env:esp32]
platform = espressif32
board = firebeetle32
framework = arduino
```

moose4lord, 2018-09-12:
You need to clean up the notifOut and notiFile instances after the notification audio has finished playing. Something like this:
```c++
if (notification && notification->isRunning()) {
    if (!notification->loop()) {
        notification->stop();
        notifOut->stop();
        delete notification;
        delete notiFile;
        delete notifOut;
        notification = NULL;
        notiFile = NULL;
        notifOut = NULL;
    }
}
```

darookee, 2018-09-12:
I do clean up the instances in my original program but didn't add it to the test code. I just modified it as suggested but that doesn't change anything.
The 'notification' sounds are very short and small and hava a variable bitrate, the 'music' has 256Kbps. Would it make a difference if I would use WAV for the small files?

moose4lord, 2018-09-12:
The mp3 bit rate is not the same as the sample rate. Sample rates of 48k, 44.1k and 32k samples/second are common for mp3 files. If you're on a Mac, you can right click on an mp3 file and select "Get Info" and it'll show you the sample rate. If there's a mismatch in sample rates between the two mp3 files, you'll get the kind of slow audio you're describing.

I tested your sketch by using an mp3 with a 48k sample rate and a short WAV file for the notification sound with a 32k sample rate, and the mp3 sound slows down when the WAV is played.

earlephilhower, 2018-09-12:
The mixer only supports everything having the same output frequency, unfortunately.  So if the first sound is set to 32khz, the next one will be output at the same rate.  It's too computationally expensive to do nice filtering on the ESP8266 where this was built to do the sample filtering, but I would be happy to get a PR with the capability from you. :)

As a start, it can either drop or replicate the samples to match the mixer output frequency (which should probably be hardcoded to 44.1k for simplicity).  No interpolation, so it won't sound quite so nice, but it's not intended to be audiophile quality, anyway.

darookee, 2018-09-12:
@moose4lord Thanks, that actually makes a lot of sense. I checked the files, the music is 44.1KHz and the effect is 24.0KHz.
Looking at the code it seems that `notification->begin(..)` tells the mixer and than the I2C out which sample rate it should use, having the slower rate it slows down the output. I added `musicOut->SetRate(44100)` after cleaning up the notification stuff and that sped up the audio again. I guess I'll redo my audio files, though, that probably sounds nicer... :-)

@earlephilhower I'd love to make a fix for this, unfortunately this is my first week in C/C++ so I'm not sure if I'll be able to do that. I will close the issue for now and see if I can get anything done that would be worth creating a PR for... ;-)

closed 2018-09-12 by darookee