earlephilhower/ESP8266Audio #133: 'Decoding error 'lost synchronization'

xfdr0805, 2018-09-17:
Running for 3086 ms...
ID3 callback for: Title = 'Piano Sample'
ID3 callback for: Performer = 'Artist Goes Here'
ID3 callback for: Album = 'Album Goes Here'
ID3 callback for: Year = '2010'
Running for 4401 ms...
Running for 5830 ms...
Running for 6831 ms...
Running for 8244 ms...
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 33265'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 33266'
.
.
.
.
.
mp3 done

this is my code:
```
#include <Arduino.h>

#ifdef ESP32
#include <WiFi.h>
#else
#include <ESP8266WiFi.h>

#endif
#include "AudioFileSourceICYStream.h"
#include "AudioFileSourceBuffer.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"
#include "user_interface.h"
#include "AudioFileSourceSPIFFS.h"
#include "AudioFileSourceID3.h"
#include "AudioFileSourceHTTPStream.h"

// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.

// Enter your WiFi setup here:
const char *SSID = "D-Link";
const char *PASSWORD = "sonavox168";

// Randomly picked URL
const char *URL = "http://192.168.0.73/ci/audios/test.mp3";

AudioGeneratorMP3 *mp3;
//AudioFileSourceICYStream *file;
//AudioFileSourceSPIFFS *file;
AudioFileSourceHTTPStream *file;
AudioFileSourceBuffer *buff;
//AudioOutputI2SNoDAC *out;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;
// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
// void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
// {
//     const char *ptr = reinterpret_cast<const char *>(cbData);
//     (void)isUnicode; // Punt this ball for now
//     // Note that the type and string may be in PROGMEM, so copy them to RAM for printf
//     char s1[32], s2[64];
//     strncpy_P(s1, type, sizeof(s1));
//     s1[sizeof(s1) - 1] = 0;
//     strncpy_P(s2, string, sizeof(s2));
//     s2[sizeof(s2) - 1] = 0;
//     Serial.printf("METADATA(%s) '%s' = '%s'\n", ptr, s1, s2);
//     Serial.flush();
// }

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
    (void)cbData;
    Serial.printf("ID3 callback for: %s = '", type);

    if (isUnicode)
    {
        string += 2;
    }

    while (*string)
    {
        char a = *(string++);
        if (isUnicode)
        {
            string++;
        }
        Serial.printf("%c", a);
    }
    Serial.printf("'\n");
    Serial.flush();
}
// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void StatusCallback(void *cbData, int code, const char *string)
{
    const char *ptr = reinterpret_cast<const char *>(cbData);
    // Note that the string may be in PROGMEM, so copy it to RAM for printf
    char s1[64];
    strncpy_P(s1, string, sizeof(s1));
    s1[sizeof(s1) - 1] = 0;
    Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
    Serial.flush();
}

void setup()
{
    //将CPU频率调整到160Mhz
    //system_update_cpu_freq(SYS_CPU_160MHZ);
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(SSID, PASSWORD);
    // Try forever
    while (WiFi.status() != WL_CONNECTED)
    {
        digitalWrite(2, 1);
        //Serial.println("Connecting to WiFi...");
        delay(500);
        digitalWrite(2, 0);
    }
    Serial.println("Connected\r\n");

    //file = new AudioFileSourceICYStream(URL);
    //file->RegisterMetadataCB(MDCallback, (void *)"ICY");
    //buff = new AudioFileSourceBuffer(file, 4096);
    file = new AudioFileSourceHTTPStream(URL);
    //file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
    //buff->RegisterStatusCB(StatusCallback, (void *)"buffer");
    buff = new AudioFileSourceBuffer(file, 4096);
    //out = new AudioOutputI2SNoDAC();
    out = new AudioOutputI2S();
    // out->SetBitsPerSample(16);
    // out->SetRate(44100);
    mp3 = new AudioGeneratorMP3();
    id3 = new AudioFileSourceID3(file);
    id3->RegisterMetadataCB(MDCallback, (void *)"ID3TAG");
    mp3->RegisterStatusCB(StatusCallback, (void *)"mp3");
    mp3->begin(id3, out);
}

void loop()
{
    static int lastms = 0;

    if (mp3->isRunning())
    {
        if (millis() - lastms > 1000)
        {
            lastms = millis();
            Serial.printf("Running for %d ms...\n", lastms);
            Serial.flush();
        }
        if (!mp3->loop())
            mp3->stop();
    }
    else
    {
        Serial.printf("MP3 done\n");
        delay(1000);
    }
}
```

MY201314MY, 2019-02-15:
As well as me

stetro, 2019-11-04:
Same here, would be cool to find a solution to this :)

earlephilhower, 2020-08-09:
There were some issues with the buffer implementation at the time the bug was filed.  I suggest retrying w/current head.

closed 2020-08-09 by earlephilhower

addrian1, 2021-03-24:
I also faced a similar problem. My installation of icecast2 + ezstream doesn't always send service packets on track end or start (I didn't find a server side fighting method).

Limiting myself to my coding skills, I wrote a wretched watchdog. The method is far from pretending to solve the problem, but it may be enough for some.

`
...
void(* resetFunc) (void) = 0;
...
void StatusCallback(void *cbData, int code, const char *string) {
...
int maxByteOffset;
  sscanf(s1, "Decoding error 'lost synchronization' at byte offset %d", &maxByteOffset);
  Serial.println(maxByteOffset);
  if (code = 257 && maxByteOffset > 1000){
   resetFunc();
  }
}
...
void loop()
{
  static int lastms = 0;
  if (mp3->isRunning()) {
    if (millis()-lastms > 1000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
     }
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    Serial.flush();
    delay(1000);
    file = new AudioFileSourceICYStream(URL);
    buff = new AudioFileSourceBuffer(file, 4096);
    buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
    mp3 = new AudioGeneratorMP3();
    mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
    mp3->begin(buff, out);
  }
}

`

positron96, 2023-09-01:
Hello. I've just tried StreamMP3FromHTTP example of 1.9.7 library and experience exactly the same problem. The board is Wemos D1 mini (ESP8266)

think-based, 2023-12-15:
Hi,
I could fix the problem via these steps:
1- convert your mp3 file bitrate to 128kbs, you can do it with several free software or online apps such as https://online-audio-converter.com/
2- modify StatusCallback function:
`void StatusCallback(void *cbData, int code, const char *string) {
  if (code != 257) {
    const char *ptr = reinterpret_cast<const char *>(cbData);
    // Note that the string may be in PROGMEM, so copy it to RAM for printf
    char s1[64];
    strncpy_P(s1, string, sizeof(s1));
    s1[sizeof(s1) - 1] = 0;
    Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
    Serial.flush();
  }
}` 
I ignored error code 257 as it could happen because of mp3 file headers and metadata.
It fixed the error and plays the mp3 files smoothly !