earlephilhower/ESP8266Audio #136: ESP32 SPIFFS Partitions greater than 2MB causes buzzing on MP3 playback

NetSensors, 2018-10-04:
I am using a 16MB version of the ESP32 Wroom and when I increase the partition size above 2MB MP3 audio output to I2S from SPIFFS stops and buzzes for a second about 3 seconds after play starts. 

The length of this buzz increases with SPIFFS partition size so a 3MB partition will have a sub 0.5 second buzz hardly noticeable  to a 12MB partition with a 1 second buzz.

zachtos, 2020-07-28:
I too have this problem, I think it's a bug with no reply.

earlephilhower, 2020-07-28:
Drop SPIFFS and use LittleFS.  There's nothing the audio lib can do about it, the SPIFFS reads just take massively longer with large files and partitions.   LittleFS was close to 10x faster on file operations on the 8266.  It's probably similar on the ESP32.

closed 2020-07-28 by earlephilhower

zachtos, 2020-07-28:
Does your library support wav playback from LittleFS?   I see an example for MIDI files.

earlephilhower, 2020-07-29:
Just substitute the `AudioFileSourceLittleFS` or make your own for FatFS using the same very simple format in that `AudioFileSourceLittleFS.h` header file...about 2 lines of code.