earlephilhower/ESP8266Audio #138: Audio Output class AudioOutputI2SNoDAC set to PDM stereo output for ESP32

jgrulich, 2018-10-15:
Hi, I like to use the audio output class AudioOutputI2SNoDAC with the PDM stereo output at the pins 25,26, same as for internal 8-bit DAC, but with the much higher quality.
Where to modify you source code to switch the I2S output?

jgrulich, 2018-10-15:
As mentioned here: #33 there are is internal 8-bit DAC, but it's sound quality is poor.
Much better quality has alternative internal PDM (pulse-delta-modulation) DAC, which has the same output pins and is supported by the I2S ESP32 core lib as well.

jgrulich, 2018-10-16:
Fore those, who like to use the Internal DAC, itmay be set here:
`AudioOutputI2S(int port=0, int output_mode=INTERNAL_DAC, int dma_buf_count = 8, int use_apll=APLL_DISABLE);`
Found that the PDM DAC is not working now at present ESP32 core.

imranaalam, 2019-10-16:
@jgrulich from last four days  i am struggling with using internal DAC. on which pin are you getting output on esp32. are you using anything else than ESP32? like I2S or external DAC or transistor?

jgrulich, 2019-10-16:
@imranaalam  The output pins are 25 and 26. The sound quality of the internal DAC is poor and PDM is not working at all. See above. The only way is to use the I2S output and some external board with I2S decoder based on PCM5102.
 Due to above mentioned and many other core lib issues I'm now using ESP8285 (smaller board than ESP8266), PCM5102A decoder and 23LC1024 SPI RAM for buffer. This works well.

jgrulich, 2019-10-20:
@imranaalam I've now tested the ESP32 and found the following.
The best ESP Core now is stable 1.0.2, directly installed inside the Arduino.
Use this modifications:
To switch the I2S to internal DAC at pins 25, 26
_out_ = new AudioOutputI2S(0, 1, 8, 0);_

You may compare with PDM, simply use this at the same pins.
_out_ = new AudioOutputI2S(0, 2, 8, 0);_ 

This increase the buffer for stable playback of internet stream.
_const int preallocateBufferSize = 110*1024;   // 110kB is max. allocable size_

zeekoe, 2021-01-26:
Thanks @jgrulich! I can confirm that for me your solution (out = new AudioOutputI2S(0, 1, 8, 0);) works on an ESP32, in contrast to the unaltered PlayWavFromPROGMEM example. I connected the 1 transistor amplifier as suggested in the main readme file, to pin 25. Both PDM and I2S DAC worked this way.

jgrulich, 2021-01-26:
OK, reopened. Good idea to mention that there need to be used internal DAC, or software-simulated delta-sigma DAC instead of AudioOutputI2SNoDAC as mentioned above.

closed 2021-02-08 by jgrulich