earlephilhower/ESP8266Audio #139: Update needed for last ESP8266 core git version

jgrulich, 2018-10-21:
There was major update of the WiFi lib. in the last ESP8266 core git version which cause this audio lib not working with the HTTP and ICY stream connection. Please update the audio lib.

> D:\Arduino\projects\libraries\ESP8266Audio\src\AudioFileSourceICYStream.cpp:41:17: warning: 'bool HTTPClient::begin(String)' is deprecated (declared at D:\Arduino\projects\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:153) [-Wdeprecated-declarations]

> D:\Arduino\projects\libraries\ESP8266Audio\src\AudioFileSourceHTTPStream.cpp:40:17: warning: 'bool HTTPClient::begin(String)' is deprecated (declared at D:\Arduino\projects\hardware\esp8266com\esp8266\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:153) [-Wdeprecated-declarations]

jgrulich, 2018-11-09:
Closed as NOT solved.

closed 2018-11-09 by jgrulich

dagnall53, 2019-05-27:
Earle, 
Could you look at this please. Its not a big issue at the moment, but might confuse people. And if a later update to the HTTP library removes the function it will cause your library to fail..
I searched to see what might be happening and although I cannot see an immediate answer, it appears that the HTTP library has deprecated the bool "begin"functions and wants to use "overload" instead (??) 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h 
(Copy of relevant piece of code below) 
It may be you have to use "bool begin(WiFiClient &client, String url);" in place of "bool begin(String url)  __attribute__ ((deprecated));"  ---  but this is beyond me..

Dagnall
`/*
 * Since both begin() functions take a reference to client as a parameter, you need to 
 * ensure the client object lives the entire time of the HTTPClient
 */
    bool begin(WiFiClient &client, String url);
    bool begin(WiFiClient &client, String host, uint16_t port, String uri = "/", bool https = false);

#if HTTPCLIENT_1_1_COMPATIBLE
    // Plain HTTP connection, unencrypted
    bool begin(String url)  __attribute__ ((deprecated));
    bool begin(String host, uint16_t port, String uri = "/")  __attribute__ ((deprecated));
    // Use axTLS for secure HTTPS connection
    bool begin(String url, String httpsFingerprint)  __attribute__ ((deprecated));
    bool begin(String host, uint16_t port, String uri, String httpsFingerprint)  __attribute__ ((deprecated));
    // Use BearSSL for secure HTTPS connection
    bool begin(String url, const uint8_t httpsFingerprint[20])  __attribute__ ((deprecated));
    bool begin(String host, uint16_t port, String uri, const uint8_t httpsFingerprint[20])  __attribute__ ((deprecated));
    // deprecated, use the overload above instead
    bool begin(String host, uint16_t port, String uri, bool https, String httpsFingerprint)  __attribute__ ((deprecated));
#endif`

earlephilhower, 2019-05-27:
Yeah, I was involved in that PR significantly.  It's a minor tweak, but with the 2.5.1/.2 Arduino release I didn't have much free time for the audio project.  Let me throw together a change...

earlephilhower, 2019-05-27:
Actually, it looks like I fixed that already.  @dagnall53, where are seeing the deprecated warning?

See the commit: https://github.com/earlephilhower/ESP8266Audio/commit/2d8a97767ce1e9ecec7494cb751a43b3d2fa872b

dagnall53, 2019-05-28:
I still had 1.1.3 Sorry. I have now manually updated to 1.1.4.