earlephilhower/ESP8266Audio #140: Noise instead of MP3 or WAV content from SD

Flos, 2018-10-22:
Hello, I try to get the mp3 music from an ESP32 (Nodemcu-32s) from a SD card to a LEORY CJMCU-1334 UDA1334A DAC

The ID3 Tags are detected, and the SD file system list all files well, I can read and write to the SD. 
The ESP8266SAM lib works, and "TTS" delivers speech to my speakers with the same wiring.
But if I try to play a WAV or MP3 file I get only noise from the speaker. 

I use: 
AudioOutputI2S
AudioFileSourceSD
AudioFileSourceID3

and tried:
AudioGeneratorMP3
AudioGeneratorMP3
AudioGeneratorWAV.h

any ideas, where I can start digging to resolve the issue?

kindzhon, 2018-10-23:
when u upload firmware,u can try change the **CPU Frequency:160MHz.**

Flos, 2018-10-24:
Ow, I thought the ESP32 is running by default on 240 Mhz and the esp8266 is running by default on 80Mhz. Somehow I thought that only the esp8266 needs this setting. 

I will try that today, @kindzhon thanks for this suggestion!

Flos, 2018-10-24:
I had no success with changing the CPU Frequency. I don't hear any difference with 160MHz or 240MHz.

I use platfromio to build my project, and I uploaded "Minimal" version of my project which does not work. It expects a 1.MP3 and 2.MP3 file on the root of the sd card.

[https://github.com/Flos/ESP32-MP3](https://github.com/Flos/ESP32-MP3)

earlephilhower, 2018-10-24:
For the ESP32 you don't get to change the CPU frequency, it's 240MHz IIRC no matter what.  On the ESP8266 you need the extra oomph so need to run at 160MHz (80MHz is the default IDE speed).

Please check your DAC, there are some which only support fixed output frequencies that maybe the MP3 is not running at.  The fact that you're getting sound using the other samples means the wiring is probably fine.

Flos, 2018-10-25:
I found my issue after testing with AudioOutputI2SNoDAC.h and the transistor... getting the same results. 
In short: **dont use delay in the loop** when you want to play a file from the SD.
  
On every loop I realized that there is another tone coming from the speaker on every loop. 
Thanks for the nice lib.

closed 2018-10-25 by Flos

imranaalam, 2019-10-16:
@flos i have tried your ESP32-MP# code using SD card its working fine for ID3. on which pin on ESP32 are  you getting the output.  what does your schematic look like?