earlephilhower/ESP8266Audio #143: ESP-Wroom-32 constant "lost synchronization"

DerMega, 2018-11-01:
Hi,

When I try to user your StreamMP3FromHTTP_SPIRAM example I get constant a lost sync error. No matter what stream I choose. What can cause this?

Stream-example: http://78.31.65.20:8080/dance.mp3
Serial output:
`Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
...Connecting to WiFi
Connected
SPI RAM buffer size: 262144 Bytes
+0 0x3ffb1f38
Running for 5460 ms...
Buffering...
METADATA(ICY) 'treamTitle' = 'Pete Bellis & Tommy - Gonna Do It (Desusino Boys & Larissa Jay '
Filling Done !
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 195073'`

thx in advance :)

taylorcoffelt, 2018-12-01:
Getting this behavior as well on ESP32 with [this code](https://github.com/anton-b/M5Stack_ESP32_radio/blob/master/ESP32_radio.ino ) on the M5Stack ESP32 platform. There's a ton of popping and stuttering before it gives up on the stream. (about a minute in)

earlephilhower, 2020-08-09:
There were some issues with the buffer implementation at the time the bug was filed.  I suggest retrying w/current head.

closed 2020-08-09 by earlephilhower

playground, 2021-03-06:
I'm getting the similar behavior on M5Stack ESP32.

STATUS(buffer) '3' = 'Buffer underflow'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 7048'
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(buffer) '3' = 'Buffer underflow'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 7049'
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(buffer) '3' = 'Buffer underflow'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 7050'
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(buffer) '3' = 'Buffer underflow'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 7051'
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(buffer) '3' = 'Buffer underflow'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 7052'
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(buffer) '3' = 'Buffer underflow'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 7053'

think-based, 2023-12-15:
Hi,
I could fix the problem via these steps:
1- convert your mp3 file bitrate to 128kbs, you can do it with several free software or online apps such as https://online-audio-converter.com/
2- modify StatusCallback function:

```

void StatusCallback(void *cbData, int code, const char *string) {
  if (code != 257) {
    const char *ptr = reinterpret_cast<const char *>(cbData);
    // Note that the string may be in PROGMEM, so copy it to RAM for printf
    char s1[64];
    strncpy_P(s1, string, sizeof(s1));
    s1[sizeof(s1) - 1] = 0;
    Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
    Serial.flush();
  }
}

```

I ignored error code 257 as it could happen because of mp3 files headers and metadata.
It fixed the error and plays the mp3 files smoothly !