earlephilhower/ESP8266Audio #144: MP3 stops (sometimes) too soon

TheEris, 2018-11-10:
Hello,
I don't know if its a bug in the library or just bad mp3s, but on some mp3s, the sounds end too soon. About ~0.5s too soon. 

I'm using this library for an Announcer for Home Assistant, that generates MP3 from google talk and send URL link via MQTT to the ESP32 to play it.

I'm using Adafruit MAX98357A for DAC and Node ESP32-S (but this problem happens on ESP8622 D1 mini/pro too). Arduino IDE 1.8.5

Here is the code I'm using
```

#include <PubSubClient.h>

#ifdef ESP32
#include <WiFi.h>
#include <WiFiMulti.h>
#else
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#endif

#include <AudioFileSourceHTTPStream.h>
#include <AudioFileSourceBuffer.h>
#include <AudioGeneratorMP3.h>
#include <AudioOutputI2S.h>

#ifdef ESP32
WiFiMulti wifiMulti;
#else
ESP8266WiFiMulti wifiMulti;
#endif

WiFiClient espClient;
PubSubClient client(espClient);

AudioGeneratorMP3 *mp3 = nullptr;
AudioFileSourceHTTPStream *stream = nullptr;
AudioOutputI2S *out = nullptr;
AudioFileSourceBuffer *buff = nullptr;

const char* mqtt_server = "XXX.XXX.XXX.XXX";

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  wifiMulti.addAP("XXXX", "XXXX");
  Serial.println("Connecting Wifi...");

  while (wifiMulti.run() != WL_CONNECTED) {
    Serial.print(".");
    delay(250);
  }

  Serial.println(" WiFi connected!");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void loop() {
  bool wifiIsConnected = WiFi.isConnected();
  if (!wifiIsConnected) {
    mp3->stop();
    delete mp3;
    mp3 = nullptr;
  }

  if (mp3 && mp3->isRunning()) {
    if (!mp3->loop()) {
      mp3->stop();
      delete mp3;
      mp3 = nullptr;
    }
  }

  if (!client.connected()) {
    reconnect();
  }

  client.loop();
}

void playAudio(char* URL) {
  if (!out) {
    out = new AudioOutputI2S();
    out->SetOutputModeMono(true);
  }
  stream = new AudioFileSourceHTTPStream(URL);
  buff = new AudioFileSourceBuffer(stream, 1024 * 4);
  //buff = new AudioFileSourceBuffer(stream, preallocateBuffer, preallocateBufferSize);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(buff, out);
  if (!mp3->isRunning()) {
    Serial.println(F("Unable to play MP3"));
    mp3->stop();;
    delete mp3;
    mp3 = nullptr;
  }
}

void callback(char* topic, byte * payload, unsigned int length) {
  Serial.print(F("Message arrived ["));
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
  playAudio((char *)payload);
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    if (client.connect("Annoucer")) {
      Serial.println("connected");
      client.subscribe("XXXXXX");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
```

treii28, 2018-11-28:
I'm running into this also playing short clips. It doesn't seem to be specific to the length of any given file, at least not per what I can tell. I have a number of short files I want to use on this project and some play fine, some cut off about a half-second too soon.

Let me know if there's anything I can do to help as I think your library is great and I'd like to help any way I can. (I'll play with .wav files tomorrow as I have both versions of these clips. I'll see if any of those are behaving in a similar fashion) I have also verified the files that were clipping do not clip on my other platforms by removing the SD card and trying them through a couple of different players. (all of the files I am using are encoded with lame at 22050/16/48k/mono )

awilkie1, 2019-05-24:
Was there a conclusion to this ? Still getting clicking at the end of clips?

earlephilhower, 2020-05-11:
Latest commit takes care of the terminal packet loss.

closed 2020-05-11 by earlephilhower