earlephilhower/ESP8266Audio #149: hanging value on wav files?

treii28, 2018-11-29:
I was running into the issue where some mp3 files would drop the last 1/2 second or so, so I switched to using wav files but now I'm seeing another issue on some files. To be fair, I may to back and edit those files again as I suspect what may be part of the problem in that the file does not end on pure silence. (I'm cutting sections out of an audio book so I'm not always getting silence at the end of a clip)

Since this entails a number of short clips that amount to sound effects, what I think is happening is that it's hanging with the value last sent after the file completes and what I hear is a oscillating noise depending on the value last sent. Is there any way to make sure the ->stop() function 'silences' the DAC upon the completion of playing a file?

I played with adding the following to the stop() function in the AudioOutputI2s::stop() and it seems to silence the dac:

    int16_t silence[2] = {0, 0};
    ConsumeSample(silence);

I only noticed it because I was already playing with getting the average abs value of both channels and was displaying the result as the output played to give me an 'envelope' value for animating parts of my project based on the playing sound volume. Only sounds that don't end on '0' produce this consequence. I can upload one as an example, but if you create any sound in a sound editor that doesn't end in silence, you should be able to repeat the effect.

treii28, 2018-11-29:
As an aside, wouldn't it also be a wise idea to initialize the 'running' values in the various generators as 'false'? I was getting a guru meditation reboot when I was trying to check wav->isRunning() in a test.

Maigre, 2019-01-30:
Hi,
i got the issue too: with I2S on ESP32, stopping MP3 is clean, 
but stopping WAV let values hanging.

To clean i2s on ESP32, we must call i2s_zero_dma_buffer,
which is called in AudioOutputI2S::stop()

AudioGeneratorMP3::stop() does call output->stop();
BUT it's missing in AudioGeneratorWAV::stop();

So the fix should be to add 
`output->stop();`
at line 46 in **AudioGeneratorWAV.cpp**

i'll test and confirm that it is ok.
I am not very familiar with PR, but i'll give it a try if this is fixing the issue.

Best,

PR #174 (2019-04-12): Call output->stop() when stopping generators
When using I2S on ESP32, i2s_zero_dma_buffer must be called when stopping playback to avoid hanging sound. This was implemented in AudioOutputI2S.cpp stop() method, but the call to this method from generators has only been added in the MP3 generator and not in other generator.

This PR adds a call to output->stop() in every generator stop() method except for MP3 generator which already had this call.

This PR fixes #149 

Best,
closed 2019-04-12 by earlephilhower