earlephilhower/ESP8266Audio #150: Bullet Sound Background with NoDAC

jafarkazemi, 2018-11-29:
Hi everybody,
I am new in esp8266. Using NoDAC, I get noisy music with a background periodic bullet sound. I use 2N3904 as the proposed schematic and 40 Ohm headset speaker. The settings is shown in the figure below:
[image]
My code:

> #include <Arduino.h>
> #include "AudioFileSourceSPIFFS.h"
> #include "AudioGeneratorMP3.h"
> #include "AudioOutputI2SNoDAC.h"
> 
> AudioGeneratorMP3 *mp3;
> AudioFileSourceSPIFFS *file;
> AudioOutputI2SNoDAC *out;
> void setup()
> {
>   Serial.begin(115200);
>   delay(1000);
>   SPIFFS.begin();
>   file = new AudioFileSourceSPIFFS("/1.mp3");
>   out = new AudioOutputI2SNoDAC();
>   mp3 = new AudioGeneratorMP3();
>   mp3->begin(file, out);
> }
> 
> void loop()
> {
>   if (mp3->isRunning()) {
>     if (!mp3->loop()) mp3->stop(); 
>   } else {
>     Serial.printf("mp3 done\n");
>     delay(1000);
>   }
> }

The reference mp3 and the received sound (recorded) is attached to be compared.
Anyone having an idea about my problem? 
Thanks 
[Sounds.zip](https://github.com/earlephilhower/ESP8266Audio/files/2629944/Sounds.zip)

dmagyar, 2019-02-23:
I have the same experience, this is with our without the transistor (I used an external amplifier for w/o transistor testing). I am running the vanilla example with pno-cs.mp3.

jafarkazemi, 2019-02-24:
I used transistor, but the problem still remained. After a couple of days, I found that the noise begins to appear on VCC at the line "_out = new AudioOutputI2SNoDAC()_;" even without playing the sound.

Jarvis785, 2021-01-30:
Hi @dmagyar & @jafarkazemi 
even I'm facing similar issue when I'm playing same mp3 in loop I keep hearing a click noise.
@earlephilhower Any help will be much appreciated !

`void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop())
    {
      mp3->stop();
      delete mp3; delete id3; delete file; delete out; // free up memory !
      file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
      id3 = new AudioFileSourceID3(file);
      id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
      out = new AudioOutputI2SNoDAC();
      mp3 = new AudioGeneratorMP3();
      mp3->begin(id3, out);
    }
  }
}`