earlephilhower/ESP8266Audio #153: Pause and Resume MP3

jenshackel, 2018-12-01:
Sorry bothering you again with a silly beginners question: I go the mp3 running (from SD-Card) and like to implement a pause/resume functon. As there is no such mp3->pause I am very clueless right now. 

Thank you very much for your support an this really cool library!

sir66, 2018-12-23:
I use this code:

file = new AudioFileSourceSD(name);
id3 = new AudioFileSourceID3(file);
id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
mp3 = new AudioGeneratorMP3();
if (!(mp3->begin(id3, out))) {
   Serial.println("Player error!");
    return; 
}
out->SetGain(vol);

while (mp3->isRunning()) {
  if (Serial.available()) {
    if (Serial.read() == 'p') {
      if (paused) {
        paused = false;
        out->begin();
      } else {
        out->stop();
        paused = true;
      }
    }
  }
  
  if (!paused)
    if(!mp3->loop()) mp3->stop();
}

jenshackel, 2018-12-29:
@sir66 
Thank you very much for your reply. I'll look into that and try to add it to my sketch!
cheers & a very happy 2019!

closed 2021-06-09 by jenshackel