earlephilhower/ESP8266Audio #154: Trigger Wav Sample

Gilgahex, 2018-12-07:
I'm trying to trigger a wav playback on an ESP32 when an analog input has reached a certain threshold.

The playback needs to be in loop() because I want the sample to play at every instance this threshold is reached

I was unable to get the PlayWAVfromPROGMEM example working, but I got the MixerSample working and have been building off of that.

Roughly my code looks as follows, 

```#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"
#include "AudioOutputMixer.h"

#include "beat.h"

AudioGeneratorWAV *wav[2];
AudioFileSourcePROGMEM *file[2];
AudioOutputI2S *out;
AudioOutputMixer *mixer;
AudioOutputMixerStub *stub[2];

...

void setup(){
  file[0] = new AudioFileSourcePROGMEM(beat, sizeof(beat) );
  out = new AudioOutputI2S();
  mixer = new AudioOutputMixer(32,out);
  stub[0] = mixer->NewInput();
  stub[0]->SetGain(0.4);
  wav[0] = new AudioGeneratorWAV();
  //Debug to ensure proper playback
  wav[0]->begin(file[0], stub[0]);
}

void loop(){
//Read in Data
if ( analogRead(36) > threshold ){
   wav[0]->begin(file[0], stub[0]);
  
   if (wav[0]->isRunning()) {
      if ( !wav[0]->loop() ) {  wav[0]->stop(); stub[0]->stop();  }
    }
}
```

I hear the first wav playback from setup(), but nothing from loop(), I also verified that my threshold has been set correctly.

I've determined that without the secondary isRunning() loop the playback fails

tablatronix, 2019-01-15:
I have the same issue, it plays only once.
Does out have to be reset ?

tablatronix, 2019-01-15:
stop(), closes the file ( or pointer )
you have to reinit or open() the source each time

eg.
```CPP
// file->open( viola, sizeof(viola) );
file = new AudioFileSourcePROGMEM( viola, sizeof(viola) );  
```

closed 2020-08-08 by earlephilhower