earlephilhower/ESP8266Audio #158: PIO compile errors

tablatronix, 2019-01-15:
esp8266/staging
PlayWAVFromPROGMEM

anyone have issues with compiling this in platformio ?

```
In file included from /Users/.../.platformio/lib/ESP8266Audio_ID1964/src/AudioFileSourceICYStream.h:31:0,
from /Users/.../.platformio/lib/ESP8266Audio_ID1964/src/AudioFileSourceICYStream.cpp:21:
/Users/.../.platformio/lib/ESP8266Audio_ID1964/src/AudioFileSourceHTTPStream.h:28:33: fatal error: ESP8266HTTPClient.h: No such file or directory
```

```
In file included from /Users/.../.platformio/lib/ESP8266Audio_ID1964/src/AudioFileSourceSD.cpp:21:0:
/Users/.../.platformio/lib/ESP8266Audio_ID1964/src/AudioFileSourceSD.h:25:16: fatal error: SD.h: No such file or directory
```

I have no issues with dependency linking these in other projects, the files exists in framework.

Ideas?

mattdavis90, 2019-04-28:
An alternative is adding
```
lib_ldf_mode = deep
```
to you ```platformio.ini```.

sauloonze, 2019-05-03:
Thanks @mattdavis90. That solved for me!

jholthusen, 2020-03-27:
just for people like me finding this as a "documentation": The config goes into int env section, like this:
```
[env:esp32dev]
lib_ldf_mode = deep
lib_deps = 
	ESP8266Audio
	
```
if you still see issues, delete ./.pio/build directory to avoid confusion with "old" configs or libraries.

closed 2020-08-08 by earlephilhower