earlephilhower/ESP8266Audio #159: SOLVED - UPDATE - Intermittent buzzing sound while running WebRadio

balimansky, 2019-01-16:
All - last spring I posted about an intermittent buzzing sound I had experienced when using the WebRadio example on an ESP-32.  After taking a break from this project I dusted off my old hardware and put some new energy into resolving this.  After a few late nights I think I have figured out an important, self inflicted, issue.

After a lot of experimentation, I realized that when streaming an audio feed from the web I had not been careful to select "stations" that were running at 128kbps.  I believe that trying to stream stations which were broadcasting at higher speeds had the unintended effect of creating a loud buzzing and ultimately a error which stopped the feed altogether.

By correcting this and focusing on stations that were broadcasting at 128kbps I have not had the issue again.

Any color on the bitrate limitation of the software/hardware would be of interest to me.

Thank you all again!  Great stuff!

Ben

eqsone, 2019-03-19:
Could you please try if this one plays on your setup: http://listen.one.hiphop/live.mp3

According to iTunes this is at 128 kBit/s, 44,100 kHz but unfortunately I can't get a tone out of it - at least not with using AudioFileSourceICYStream.

The aac variant also doesn't do, yet: http://listen.one.hiphop/live

earlephilhower, 2019-04-13:
128kbit stereo is probably the limit on the 8266, but it's hard to quantify since the internet and buffering also come into play.  From SD card or ROM, I think I've seen 192kb play OK (not SPIFFs, that's too slow).

As for AAC, there isn't enough free RAM in the 8266 to decode it.  AAC works on larger buffers than MP3 by 2x-3x, and it just doesn't fit.

closed 2019-04-13 by earlephilhower