earlephilhower/ESP8266Audio #160: ESP8266Audio not in Arduino Library Manager

terryspitz, 2019-01-17:
on: Arduino 1.8.8 gui, on Windows.
Should you be listed in the Arduino Library Manager?  it's not showing for me.

per1234, 2019-01-17:
The library is not listed in Arduino Library Manager because it has never been submitted for inclusion.

In order for this library to be added to the Arduino Library Manager index, the following would need to happen:
1. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your [library.properties file](https://github.com/earlephilhower/ESP8266Audio/blob/master/library.properties#L2).
1. Open an issue in the [arduino/Arduino repository](https://github.com/arduino/Arduino) requesting to be added, with the URL of this repository.

If this is something that is considered desirable, I'd be happy to help by doing the second step, but I can't do the first.

Library Manager offers an easy way to install and update the library. The Library Manager indexer job automatically picks up any new tags/releases within an hour so after the library is added to the index there is no additional work that needs to be done. The Arduino IDE will display a notification (if enabled in preferences) when any installed library has a newer version available from Library Manager.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

terryspitz, 2019-01-19:
>    2. Open an issue in the arduino/Arduino repository

It's a great library and I'm sure this would help people find it.

TheeCaptain, 2019-05-10:
It would also help people install it. I am having difficulty with dependencies. It would be great if I could just use the Arduino IDE to install this on my ESP32 and test out the library.

dagnall53, 2019-05-28:
Add me to the list of people asking for this as well!
This is a great library, with excellent support.  I missed an update because I was only looking at the library manager, and not checking here.

earlephilhower, 2020-08-10:
https://github.com/arduino/Arduino/issues/10623

earlephilhower, 2020-08-20:
Done, yay!

closed 2020-08-20 by earlephilhower