earlephilhower/ESP8266Audio #161: Check for File presence in SPIFFS

dagnall53, 2019-01-21:
Earle, 
I am sure this is trivial, but I am trying to put in checks that a File actually exists before allowing my sound code to run. 
I tried something like this 
```
if (AudioFileSourceSPIFFS.open(wavfilename)){
           file[Channel]= new AudioFileSourceSPIFFS(wavfilename);}
else {DebugSprintfMsgSend( sprintf ( DebugMsg,"SFX No such file name"));     }
```

But I get 
      `"expected primary-expression before '.' token"`

Is there an easy way to use the file open Boolean that seems to be in your AudioFileSourceSpiffs.cpp 
`(bool AudioFileSourceSPIFFS::open(const char *filename) ` ? 

Reason for doing this is that the code seems to accept non present SPIFFS filenames, but then does not detect end of file, so "plays" non stop.. (thus stopping me starting a new file sound play).
I'm trying to update my Rocrail sound stuff to ESP32 and wanted to put in some checks to make it more bulletproof..
Huge thanks
Dagnall

dagnall53, 2019-01-26:
I am now using this code in case anyone want to copy or improve..
`
bool FileExists(const char *wavfilename){
 File TestObj;
  if (!(TestObj= SPIFFS.open(wavfilename,"r"))){ 
    #ifdef _SERIAL_Audio_DEBUG
        DebugSprintfMsgSend( sprintf ( DebugMsg, " SFX File not found %s",wavfilename));
    #endif
        malloc(TestObj); 
        return false; 
        }
  malloc(TestObj); 
return true;
}`

dagnall53, 2019-02-16:
Opened again as The code above only works for ESP8266 NOT esp32.. 
(although my esp32 does seem to be stable even if I open a non existent file..!)

earlephilhower, 2020-05-11:
````
if (AudioFileSourceSPIFFS.open(wavfilename)){
           file[Channel]= new AudioFileSourceSPIFFS(wavfilename);}
else {DebugSprintfMsgSend( sprintf ( DebugMsg,"SFX No such file name"));     }
````
But I get
"expected primary-expression before '.' token"

You've got a C++ syntax problem.  You need to make an object of the class, then use that variable to call open and check the return value:
````
AudioFileSourceSPIFFS f;
if (!f.open(wavfile)) ....
````

closed 2020-05-11 by earlephilhower