earlephilhower/ESP8266Audio #164: Audio Hiss with UDA1334A

danbicks, 2019-01-27:
Hi,

Great library, I have hooked up a UDA1334A I2S dac and have a very loud hiss.

If I hold the 3 data cables to the dac the hiss disappears and audio is heard.
Looked on the net and someone mentioned audio_dither?

Can anyone help on this?

Big thanks

Dans

danbicks, 2019-01-27:
Found the issue for anyone else:
Pin SCLK on the board should be taken high, use a 10k resistor, this cures the background hiss completely and all works.

Cheers

Dans

earlephilhower, 2019-02-04:
Yes, it seems each codec has its own requirements.  You really need to read the datasheet carefully, which it looks like you did, to make sure unused inputs are tied in a safe stae.

closed 2019-02-04 by earlephilhower