earlephilhower/ESP8266Audio #165: ESP32 Webradio

danbicks, 2019-01-27:
Hi,

I have tried the example Webradio example on the ESP32, all compiles and builds when the unit restarts I access the url through chrome browser, serial monitor shows sending HTML so request has been made but shows not responding web page?

Has anyone had a similar issue?

Cheers

Dans

2ni, 2019-02-25:
Yup, have similar issue. Webserver returns ERR_EMPTY_RESPONSE. Seems that HandleIndex is not working properly?
Btw, did you try with /index.html

danbicks, 2019-02-25:
Hi 2ni,

Same for me, I can't believe 29 days later no one has replied to my first post!

Seems like this development may be dead...

Let me know if you have some success, that would be cool.

Cheers

Dans

waipeinan, 2019-07-02:
I seem to have this issue too..... are we really the only ones?

PLdeSousa, 2019-07-17:
I have the same problem.

DerMega, 2019-09-26:
Same here. Ist there a solution meanwhile?

jgrulich, 2019-10-20:
I've the same issue till now.
The only core where works the web is ver. 0.0.1 dated to 2018-07-04. But you need to use the old version on the ESP8266Audio as well, otherwise it results in a lot of compilation errors.
I've tested all ESP32 stable cores with the latest ESP8266Audio. The 1.0.0 has compilation error, the rest compiles, but only 1.0.1 and 1.0.2 are working with audio. The 1.0.2 has best memory handling now. There may be up to 110kB of buffer.
As the html requests are handled correctly, the config may be done with sending of direct html commands. But I've not bring this web.h server to send any response. As it works with the ESP82 core, there need to be something with the wrong lib interpretation.

Cocksparrow, 2020-01-13:
Thanks jgrulich!  Same here.  I've rolled back to 1.0.2 and the audio works.  1.0.0 doesn't compile and 1.0.1 compiles but then doesn't stream audio.  I thought I had lost my mind but you reminded me I updated the boards about the same time as everything stopped working.  Any advice anyone has on an alternative example with a working web interface would be gratefully received.

Derek-K, 2020-12-09:
Update here... Just tried 1.7.0. 

TL;DR Web interface still doesn't work with ESP32, but it is OK on ESP8266

So far I have spent over an hour trying to debug the code under ESP32 without luck, But the same code (web.cpp) works just fine on ESP8266.

In particular, I suspect it has something to do with the WebPrintf inWebHeaders(). I even added client->flush(); at the end, still, nothing was sent. Also interestingly WebError() is almost the same and works under ESP32.

ITguy-Lukas, 2022-01-02:
Recently I ran over this problem - and pinpointed it to the WebReadRequest call before the if/then/else statement for client request. As stated elsewhere client.flush() is to be used to erase all pending data in the reception buffer. So I replaced call to client->flush() by <code> while (client->available()) {
    client->read();
}</code>

szybkijanek, 2023-04-17:
Just comment line 181 in web.cpp //client->flush(); // Don't need anything here.... This works for me.

pentode1, 2024-02-10:
Edit:  no idea why the code blocks aren't formatting properly and I'm too tired to figure that out right now.  

After commenting out that line, the web server works but anytime I enter a stream link, the device loops, apparently for different reasons depending on the stream type.  Here's what I get:  

`Changing URL to: http://stream.radioparadise.com/mellow-128, vol=100
Before stop...Free mem=111256
After stop...Free mem=111460
Saved settings
created icystream
created buffer
created decoder
Decoder start...
+0 0x3ffcabc0
Done start new URL
MP3:ERROR_BUFLEN 0` 

or

`Changing URL to: https://stream.srg-ssr.ch/rsj/mp3_128.m3u, vol=100
Before stop...Free mem=111868
After stop...Free mem=111868
Saved settings
created icystream
created buffer
created decoder
Decoder start...
MP3 source file not open
Can't connect to URLDone start new URL
Changing URL to: https://stream.srg-ssr.ch/rsj/mp3_128.m3u, vol=100
Before stop...Free mem=111868
After stop...Free mem=111868
Saved settings
created icystream
created buffer
created decoder
Decoder start...
MP3 source file not open`

Anyone know what's going on here?  As others have noted this code works just fine on the 8266, so is this a library problem or am I doing something else wrong?  Since others seem to have made this work, I'm thinking maybe something changed in the last year?