earlephilhower/ESP8266Audio #170: Choppy output from MIDI input (even with example)

rsteck, 2019-02-06:
First, a wizard-level achievement to get this kind of functionality out of a pretty low end processor!  Color me very impressed.  I am building a music box that plays a different tune for each day of the year with configurability through a web interface and tied into an MQTT server.  MIDI playback is the key.

I'm getting choppy audio out even with the example MIDI code.  It sounds like it's processing in chunks with a lot of gaps in the audio stream.  Queuing problem?  I cannot figure out where in the code this might be failing.  The first tones seem to be pretty good and then get progressively choppier with what sounds like just a few samples being output at a time followed by underruns.  

  I've verified the AudioOutputI2S works by running AudioGeneratorWAV with great results.  I'm using the Wemos D1 Mini and have ensured that I have the latest GIT for the ESP8266 tools.  Code is below, but it's just the example code.

I would very much appreciate getting the example code to work to leverage from.

#include <Arduino.h> 
#include "ESP8266WiFi.h"
#include <WiFiClient.h>
#include <WiFiUdp.h>
#include <i2s.h>
#include <i2s_reg.h>
#include <pgmspace.h>
#include <Ticker.h>
#include <FS.h>

extern "C" {
#include "user_interface.h"
}

//char ssid[] = "SSID"; //  your network SSID (name)
//char pass[] = "SSID Pass";    // your network password (use for WPA, or use as key for WEP)

#include "AudioOutputNull.h"
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMIDI.h"
#include "AudioOutputI2S.h"

AudioGeneratorMIDI *midi;
AudioFileSourceSPIFFS *sf2file;
AudioFileSourceSPIFFS *midfile;
AudioOutputI2S *out;

void setup() {
  //WiFi.forceSleepBegin();             
  //delay(1);                               
  //system_update_cpu_freq(160);

  Serial.begin(115200);
  
  //WiFi.begin(ssid, pass);

  SPIFFS.begin();
  
  midfile = new AudioFileSourceSPIFFS("/furelise.mid");
  sf2file = new AudioFileSourceSPIFFS("/mgm.sf2");

  midi = new AudioGeneratorMIDI();
  midi->SetSoundfont(sf2file);
  out = new AudioOutputI2S();
  
  midi->begin(midfile, out); 
}

void loop() {

  if (midi->isRunning()) {
    if (!midi->loop()) {
      uint32_t e = millis();
      midi->stop();
    }
  } else {
    Serial.printf("MIDI done\n");
    delay(1000);
  }
}

earlephilhower, 2019-02-06:
Howdy,

Yes, that's kind of where things are on the MIDI.  It's not *that* compute intensive, the problem is that SPIFFS is insanely slow on large files (aka the samples file).  I have a ESP8266FastRomFS repo you could look at which does things much faster (but is less capable and robust than SPIFFS by a mile), and a PR in to the 8266 Arduino to add in LittleFS which is many X faster as well (plus is robust and tested).

Without any of those, there's not much that I can suggest.  You *might* have better luck with the samples file on an SD card, but I have not personally tried id (also I've got a new lib SD PR in the 8266 Arduino repo which may make it faster/more stable)...

The MOD player does largely the same thing, but with much smaller datasets (Amiga was 8bit IIRC, and the samples are much shorter and there are less things like reverb/etc. to stress the IO).  If you need something more stable, and you can convert from MIDI->MOD (with accompanying quality loss), that may be the best for you.

Good luck,
-EFP3

closed 2019-02-06 by earlephilhower

rsteck, 2019-02-06:
Thanks for the quick response on this.  The very slow SPIFFS has bitten me more than once, but I thought you had some optimizations on this since the WAV player is working great.  The interesting thing here is that the .sf2 file is only about 180KB so I thought that would not be such an issue.  I suspect it’s more the access pattern for each tone generator than the size of the file.

Excuse my ignorance, but what is a PR?  Is there a way to get the LittleFS without an Arduino release?   That sounds like the only option at the moment for my evil plan to work out…😊

Randy

earlephilhower, 2019-02-06:
PR is pull request.  To use them you 'd need to use the GIT version of the arduino 8266 core.  See https://github.com/esp8266/Arduino/pull/5511 

You can pull the current GIT then apply the PR with some magic that GitHub has described.  You'd still need to change the AudioFileSourceSPIFFS to use "LittleFS.open()" but that's pretty trivial.

rsteck, 2019-02-07:
Hate to bother you again, but thought I’d try loading everything into PROGMEM for MIDI processing since the soundfont file is simple and only 107K large.  This should, in theory, work given your examples, but I’m getting a compiler error and perhaps you’ve seen this same thing before.

Any idea how to get around this?

c:/users/randy/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x4000afe8 of C:\Users\Randy\AppData\Local\Temp\arduino_build_720506/MusicBoxMIDI.ino.elf section `.bss' is not within region `dram0_0_seg'

collect2.exe: error: ld returned 1 exit status

Perhaps I’m trying to do something that is known to not work, but thought I’d ask an expert.

I think your audio library for these processors is genius.

Cheers,
Randy Steck

Program listing:

#include <Arduino.h>
#include "ESP8266WiFi.h"
#include <WiFiClient.h>
#include <WiFiUdp.h>

extern "C" {
#include "user_interface.h"
}

#include "AudioOutputNull.h"
#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorMIDI.h"
#include "AudioOutputI2S.h"

#include "JonnyMIDI.h"
#include "MusicBoxRS.h"

AudioGeneratorMIDI *midi;
AudioFileSourcePROGMEM *sf2file, *midfile;
AudioOutputI2S *out;

void setup() {
  //system_update_cpu_freq(160);

  Serial.begin(115200);

  midfile = new AudioFileSourcePROGMEM(JonnyMIDI, sizeof(JonnyMIDI));
  sf2file = new AudioFileSourcePROGMEM(MusicBoxRS, sizeof(MusicBoxRS));

  midi = new AudioGeneratorMIDI();
  midi->SetSoundfont(sf2file);
  out = new AudioOutputI2S();

  midi->begin(midfile, out);
}

void loop() {

  if (midi->isRunning()) {
    if (!midi->loop()) {
      uint32_t e = millis();
      midi->stop();
    }
  } else {
    Serial.printf("MIDI done\n");
    delay(1000);
  }
}

rsteck, 2019-02-08:
Never mind.  I thought .rodata was in flash, but it’s in RAM and I’m trying to load too much.

Cheers,
Randy

From: Randy Steck
Sent: Wednesday, February 6, 2019 2:49 PM
To: 'earlephilhower/ESP8266Audio' <reply@reply.github.com>
Subject: RE: [earlephilhower/ESP8266Audio] Choppy output from MIDI input (even with example) (#170)

Thanks for the quick response on this.  The very slow SPIFFS has bitten me more than once, but I thought you had some optimizations on this since the WAV player is working great.  The interesting thing here is that the .sf2 file is only about 180KB so I thought that would not be such an issue.  I suspect it’s more the access pattern for each tone generator than the size of the file.

Excuse my ignorance, but what is a PR?  Is there a way to get the LittleFS without an Arduino release?   That sounds like the only option at the moment for my evil plan to work out…😊

Randy

earlephilhower, 2019-02-08:
Just add the PROGMEM directive to move it to ROM, like:
`const char blah[] PROGMEM = {1,2,3,4,...};`

rsteck, 2019-02-09:
Well, that worked and now the MIDI is running nicely.  Really not that much more awkward than having to use the flakey SPIFFS plug-in to get the file area loaded.

Just one more issue with the synth cutting off sustained notes on note-off, but I’ll get to the bottom of that.

Thanks,
Randy

FrankHovis, 2021-02-18:
The example file should have a note added to explain that MIDI playback from SPIFFS is effectively imposable.  I've just spent the last hour playing with settings trying to get it working...

earlephilhower, 2021-02-18:
Yeah, SPIFFS is horribly slow.  However, just replacing the SPIFFS with LittleFS makes it work almost flawlessly since it is much higher speed.