earlephilhower/ESP8266Audio #175: [Feature Request] Output direct to Bluetooth speaker

dgm3333, 2019-03-12:
Espresso have implemented bluetooth source and, since that bluetooth speakers are ubiqiutous, being able to pipe audio would eliminate the need for a separate DAC.
https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_source
"This example will download an mp3 file from HTTP and decode it into PCM format, then push it to the BT Speaker through BT Source to play it."

It would appear that tueddy has provided the code to use the ESP as an audio sink, so presumably audio via bluetooth is not too different https://github.com/earlephilhower/ESP8266Audio/issues/119

dgm3333, 2019-03-12:
Oh just found this which appears to have ported the solution to Arduino, so solves my problem. Would be great if possible with this library though...
https://github.com/dgiebert/techfest17/blob/master/A2DP_Test/A2DP_Test.ino

dgm3333, 2019-03-17:
If anyone is interested I managed to get the espresso A2DP_Source example to compile under Arduino and play static to a BT Headset: 
https://github.com/dgm3333/Arduino_A2DP_Source

earlephilhower, 2020-08-08:
This is interesting, but I don't do much on the ESP32.  If you have a (even partial) PR, throw it my way!

closed 2020-08-08 by earlephilhower