earlephilhower/ESP8266Audio #176: HTTP Streaming: Differences beyond bit- and samplerate in MPEG-Streams?!

eqsone, 2019-03-14:
Hey there, 

great work, thanks for sharing!

Noob trying to determine why some MP3 streams play flawlessly while others keep losing sync, causing real bad stuttering. Are there any additional decoding related differences besides bit- and samplerate within the streams causing this?

Following examples both seem to be 128kBit/s, 44,100kHz MPEG-streams, can someone tell what exactly causes the second one to fail? Ist this RAM related?

http://jazz.streamr.ru/jazz-128.mp3 (>plays fine)
http://listen.one.hiphop/live.mp3 (>loses sync/ bad hiccup)

Setup: NodeMCU v2 | 23LC1024 Spiram | PCM5102 DAC

eqsone, 2019-03-14:
Ok sorry, I overlooked the most obvious hint in the serial output:

Working URL:
(...)
Buffering...
METADATA(ICY) 'StreamTitle' = 'David Ruffin - Don't You Go Home'
Filling Done !
(...)

Stuttering URL:
(...)
Buffering...
METADATA(ICY) '���V#�dy��IK�S
2000
treamTit' = 'Joey Bada$$ - Big Dusty'
Filling Done !
(...)

So this points to an issue within the AudioFileSourceICYStream.cpp, right?!

eqsone, 2019-03-15:
Closed, seems to be related to this issue https://github.com/earlephilhower/ESP8266Audio/issues/124

eqsone, 2019-03-15:
According to ancient Winamp's file info, the metadata interval seem to be the reason. 8192 is ICY standard for the number of bytes between two metadata chunks, this value seems to be hardcoded within the AudioGeneratorMP3.cpp in lines 300 & 345. Obviously if this interval differs, the stream gets messed up.

(Working URL:)
Server: Icecast 2.4.0-kh8
Content-Type: audio/mpeg
Metadata interval: **8192** bytes
Stream name: jazzmmx

(Stuttering URL:)
Server: nginx/1.10.3 (Ubuntu)
Content-Type: audio/mpeg
Metadata interval: **16000** bytes
Stream name: One Love Hip Hop Radio

eqsone, 2019-03-18:
Ok, it's not the interval. This one is also at 16000 bytes and plays flawlessly: http://soulfood.stream.laut.fm/soulfood

Unicode/ASCII related?! Further investigating.. 
What does 'punt this ball for now' regarding 'bool isUnicode' actually mean?

earlephilhower, 2019-04-13:
ICY metadata decoding was actually very slow, and the hip hop website you listed is sending large metadata packets filled with random data (as far as I can see).  So it was doubly-bad.

I just threw together a different parser which works on chunks and not byte-by-byte.  It should be faster, although with the station you listed it's wasting ~20% of the BW on garbage metadata so there may still be hiccups.  You can try increasing the read buffers in the app, since the Arduino core has freed up a lot of heap since I wrote the original examples.

eqsone, 2019-04-13:
Great, thanks for your effort!

garageeks, 2019-06-28:
> You can try increasing the read buffers in the app, since the Arduino core has freed up a lot of heap since I wrote the original examples.

I'm using the latest commit on the latest ESP32 release (1.0.2), I still get a hiccup when ICY metadata is decoded, despite a 100KB buffer (tried 128KB but it can't allocate it).
This is the webradio I'm testing [link](http://icy-6.radioparadise.com:80/mp3-128)
 If I disable metadata callback it plays the stream just fine, but would be nice to get song titles. 
Do you have any advice?
Thank you

eqsone, 2019-06-28:
I gave up on this. Recently I switched to KaRadio32 instead, which plays anything you feed it with including meta tags. It‘s a whole different approach of cause, as it uses an additional VS1053B for the decoding stuff and you‘ll have to flash it using the espressif esp-idf instead of the arduino ide. 
Very well documented and highly customizable. If you‘re heading for a rock solid diy webradio rather than a basic proof-of-concept device, you might give it a try.

garageeks, 2019-06-28:
Thank you for the feedback. I tried this because I'm not familiar yet with the esp-idf. It is a great piece of software, kudos to @earlephilhower, and I liked the ability to play music by simply connecting an amp with no additional hardware, but the cracking noises are a dealbreaker.
I have a VS1053B board, will give it a try

eqsone, 2019-06-28:
I feel you. The simplicity with ESP8266Audio is quite convincing, I highly appreaciate the attempt. Yet avoiding the espressif ide if you‘re unfamiliar with it is tempting.
On the other hand, KaRadio will take the final device to a whole other level and learning about the espressif ide wasn‘t nearly as hard as suspected. It‘s really well explained within the project, just take your time to read into.