earlephilhower/ESP8266Audio #181: I2S and I2C at the same time

frederic-bonjour, 2019-04-27:
Hello,

Thanks for this great library!

I'm experiencing troubles with having both I2S (with external Adafruit DAC) and I2C working at the same time on an ESP32 dev board.
Pin 22 (DOUT) seems to be required and hard-coded in the lib. If I change it to use pin 27, this works as well. But when I use I2C at the same time (which requires pin 22), the sound is affected and becomes really awful!

Do you have any idea?
Is pin 22 strictly required to make external DAC working?

Thank you for your answers!

Regards,
Fred

frederic-bonjour, 2019-04-27:
Well, I found a solution ;)
I added pull-up resistors on SDA and SCL lines and everything works fine together: clear sound (with DOUT=27) and MCP23017 (IO expander) on I2C!

But, calling `out->SetPinout(26, 25, 27);` does not work: sound is OK, but the I2C communication seems to be broken.

I have created a pull request with my solution.

garageeks, 2019-06-26:
Awesome, I have an application with internal DAC and I2C together and it works

closed 2020-08-08 by earlephilhower

dwindler73, 2023-06-01:
This approach work perfect for me. Thanks for the information!