earlephilhower/ESP8266Audio #183: How to combine ESP2866 Audio with Wemos SD card on a Wemos Mini D1

byTheo, 2019-04-30:
After spending hours on searching how to hookUp a Wemos SD card reader in combination with the ESP8266Audio lib I can confirm I'm stuck. I can read SD contents. But the audio lib won't play my SD files. I know that the both of them interfere. But I can't find out what to change.

Help is appreciated.

btw. Love this library

ebradbury, 2019-05-30:
@byTheo Did you figure this out? I have the same setup and I'm not sure how to resolve the fact that the SD reader and DAC use GPIO15

earlephilhower, 2019-05-30:
You need to change the SD chip select to a pin that's not hardwired toe the I2S ports.  On my setup, I always use wires to a separate board, not a shield, so I use GPIO0 as the CS and all's well with the world.  Maybe the shield allows for reassigning the pin physically (jumper, short, etc.).

byTheo, 2019-05-30:
@ebradbury I wasn't. But changed tactics. It took a while, but I'm able to play MP3 files from spiffs now. If I convert them to 32 kbps with 32 khz frequency it works fine, better quality results in heavy distortion. I guess the Wemos D1 mini just isn't powerful enough. But as I only need some sound effects for my meditation timer, this setup works fine for me.

@earlephilhower the new version suppose to support physically change. The older version don't. I ordered a new version just to try. But I have to wait for them to arrive.

lulersoft, 2020-03-06:
> You need to change the SD chip select to a pin that's not hardwired toe the I2S ports. On my setup, I always use wires to a separate board, not a shield, so I use GPIO0 as the CS and all's well with the world. Maybe the shield allows for reassigning the pin physically (jumper, short, etc.).

think u,i use SD.begin(D0),it's work.

closed 2020-08-08 by earlephilhower