earlephilhower/ESP8266Audio #187: endianness on helix aac

philippe44, 2019-05-30:
Hi - I don't know if this is on purpose, but for the helix aac decoder, in assembly.h, the ARDUINO defaults to same non-accelerated function as powerpc. Problem is that my ESP32, the xtensa is using little endian, so the union U64 is in wrong order. I had decoding errors and hissing noise regulary. Moving to little endian order worked fine

```
typedef union _U64 {
	Word64 w64;
	struct {
#ifdef __XTENSA__		
		unsigned int lo32;
		signed int   hi32;
#else
		/* PowerPC = big endian */
		signed int   hi32;
		unsigned int lo32;
#endif		
	} r;
} U64;
```

earlephilhower, 2019-06-19:
Thank you for the detailed debugging!  I ported the helix AAC decoder a long time ago, and since it is only ESP32 compatible (not enough RAM on the 8266) I did a few simple tests and called it a day.

Could you make a PR for this?  Since you found it, I'd like to make sure your name is in the commit logs.

Thx
-EFP3

earlephilhower, 2019-06-21:
Fixed with your PR, thx!

closed 2019-06-21 by earlephilhower