earlephilhower/ESP8266Audio #188: What is the correct MP3 setting for playing from SPIFFS

byTheo, 2019-05-30:
I was able to get it to work, with the ESParkle project. But I can't seem to use my own MP3's. When I put them on SPIFFs and play them they sound horrible. It sounds like over distorted.

So I'm curious what software should I use, with which settings - I have a Mac Book Pro - to be able to play decent sounding sound effects?

byTheo, 2019-05-30:
Never mind got it working. Mp3 from Spiffs on a Wemos D1 mini. I used Itunes - don't ask - to convert the MP3s. The settings 32 kbps with 32 Khz sample seems to be the max. Otherwise the MP3s get distorted. I've ordere an ESP32 to see if I can juice up the sound quality.

Playback is through Adafruit MAX98357 I2S Class-D Mono Amp.

earlephilhower, 2019-05-30:
Try LittleFS instead of SPIFFS.  SPIFFS is very, very slow for file access, and it takes so much time MP3s can't play properly in this case.  You could also try the PROGMEM option if the MP3 fits within 1MB which is as fast as it gets.

closed 2019-05-30 by earlephilhower