earlephilhower/ESP8266Audio #189: white noise when reading 16 bit samples

tonysnail, 2019-06-01:
Hi Earle,

I'm trying to understand how your library writes sample values to the I2S bus. At present, the example files to play the FLAC sample, and the MOD file are working fine on my hardware.

I have a PCM5102 wired to an ESP32 board.

My example code reads a 16 bit sinewave table, 1024 samples in size, and sends it to the I2S left/right audio buffer using the consumeSample() method. 

I'm expecting to hear a high-pitch sinewave - however, I'm only getting white noise, and I've been unable to understand why.

Here is my source code:
`#include <Arduino.h>
#include "AudioFileSourcePROGMEM.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2SNoDAC.h"
#include "sine.h"

AudioGeneratorWAV *wav;
AudioFileSourcePROGMEM *file;
AudioOutputI2S *out;

int16_t pos  = 0;
int16_t sineSize  = sizeof(sine);

void setup() {
  Serial.begin(115200);
  out = new AudioOutputI2SNoDAC();
}

void loop() {
  uint16_t sampleread[2];   // read samples from table (unsigned values)
  int16_t  sample[2];

  bool bufferFull = 0;
  do
  {
    pos++;
    if (pos > sineSize) pos = 0;

    // read from sine table (unsigned values from 0 to 65535)
    sampleread[0] = sine[pos];
    sampleread[1] = sine[pos];

    // zero samples and put into signed ints
    sample[0] = (int16_t)sampleread[0] - 32768;
    sample[1] = (int16_t)sampleread[1] - 32768;

    // send samples to I2S buffer
    bufferFull = out->ConsumeSample(sample);
  } while (bufferFull == 1); 
}`

Many thanks,
Carl Hudson

meltdown03, 2019-06-11:
Could it be that you include the AudioOutputI2SNoDAC.h header? Then you declace *out as AudioOutputI2S, but in your setup function you create a new instance of the AudioOutputI2SNoDAC class.

earlephilhower, 2019-06-25:
That is your problem.  You're asking the 8266 to send out a sigma-delta "analog" bitstream to the DAC, when you want to just send the raw bits.  Use AudioOutputI2S() instead and it should send the proper stream to your DAC.

closed 2019-06-25 by earlephilhower