earlephilhower/ESP8266Audio #194: ESP32 problems playing WAV from SD card

AidanTek, 2019-06-26:
Hello, I am sure many have done this before me - I am trying to play back a wav file from an SD card. I am trying to adapt the example from the sparkfun website, which uses interrupt triggers to playback mp3s - I have bypassed this and I just want it to play! Please take a look at the code below and tell me where I am going wrong :) 

I am using the Adafruit Huzzah32 - I had to change i2s pins in AudioOutputI2S.cpp - I verified this works using one of the PROGMEM examples.

I have tested the SD card seperately which is happily printing the directory contents

I get three errors that say:
* E (1191) vfs_fat: open: no free file descriptors
* AudioGeneratorWAV::ReadWAVInfo: failed to read WAV data
* AudioGeneratorWAV::begin: failed during ReadWAVInfo

Code below:

```cpp
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include "driver/i2s.h"
#include <SD.h>

//Initialize ESP8266 Audio Library classes
AudioGeneratorWAV *wav;
AudioFileSourceSD *file;
AudioOutputI2S *out;

volatile bool playing = 0;
volatile byte loadTrack = 0;

void setup() {  
  Serial.begin(115200);

  out = new AudioOutputI2S();
  wav = new AudioGeneratorWAV();

  delay(1000);
  Serial.print("Initializing SD card...");
  if (!SD.begin(33))
  {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
  delay(100);
}

void loop() {
  if(!playing) {
    loadTrack = 1;
  }
  
  if(loadTrack) { // Load the track we want to play
    //Stop the current track if playing
    if(playing && wav->isRunning()) wav->stop();

    file = new AudioFileSourceSD("/BUBBLES.WAV");

    out -> SetGain(0.08); //Set the volume
    wav -> begin(file,out); //Start playing the track loaded
    loadTrack = 0;
  }

  if(playing && wav->isRunning()) {
    if (!wav->loop())
    {
      wav->stop();
      playing = 0;
      Serial.println("Stopped");
    }
  }
}
```

AidanTek, 2019-06-26:
I noticed that I missed out the 'playing' logic lock when I removed the interrupt service routine - I have just added it after the begin() function, now I just get one error:

* AudioGeneratorWAV::begin: file not open

I need a break now, current code is:

```cpp
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include "driver/i2s.h"
#include <SD.h>

//Initialize ESP8266 Audio Library classes
AudioGeneratorWAV *wav;
AudioFileSourceSD *file;
AudioOutputI2S *out;

volatile bool playing = 0;
volatile byte loadTrack = 0;

void setup() {  
  Serial.begin(115200);

  out = new AudioOutputI2S();
  wav = new AudioGeneratorWAV();

  delay(1000);
  Serial.print("Initializing SD card...");
  if (!SD.begin(33))
  {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
  delay(100);
}

void loop() {
  if(!playing) {
    loadTrack = 1;
  }
  
  if(loadTrack) { // Load the track we want to play
    //Stop the current track if playing
    if(playing && wav->isRunning()) wav->stop();

    file = new AudioFileSourceSD("/BUBBLES.WAV");

    out -> SetGain(0.08); //Set the volume
    wav -> begin(file,out); //Start playing the track loaded
    playing = 1;
    loadTrack = 0;
  }

  if(playing && wav->isRunning()) {
    if (!wav->loop())
    {
      wav->stop();
      playing = 0;
      Serial.println("Stopped");
    }
  }
}
```

AidanTek, 2019-07-03:
Without changing the above code, it turns out that sometimes the SD Card is mounted successfully and sometimes it isn't. Increasing the delay() after SD.begin() seems to improve the chance. I have audio playing now, but the quality is very bad - very choppy. I am playing a 16bit WAV at 44.1kHz. 

As mentioned above, I had to change the i2s pins in the library itself as the Feather HUZZAH32 doesn't break out the pins mentioned - is there a chance that this is causing grief? It seemed to work ok with the library example playing a WAV from PROGMEM

I am going to try an MP3 instead as I am wondering if the data transfer speed is the issue - any suggestions would be very welcome.

AidanTek, 2019-07-03:
Last comment from me, MP3 at 160kbs works a lot better, but is still full of noisey artifacts - I think this must be to do with transfer speed. There is a level shifting chip on the SD card module which doesn't need to be there - I am going to try bypassing it to see if that helps.

earlephilhower, 2019-07-03:
The voltage converter wouldn't affect the speed.  It might cause things to fail, but it's not adding any clock cycles latency.

You could try upping the SPI speed.  For the 8266 I've added this to the standard SDFS libraries, but for the -32 I'm not sure the exact sequence you'd need.  It should be easily documented and performed ad/with the SD.begin() call.

AidanTek, 2019-07-03:
Thanks @earlephilhower - I have literally just been reading up about this, it's about time to give it a rest for today so I will report back soon. I will contribute an example once I have cracked this if you are happy for me to do so

AidanTek, 2019-07-08:
Hello again - I had a play this afternoon and through some testing I found that I could change the file read/write speed with the sd.begin() command using the ESP32 SD example.

```cpp
SD.begin(33,SPI,8000000);
```

I am using a SanDisk Ultra Class 10 which should be pretty good. I don't get much of an improvement beyond 8000000. However, while this is double the speed the library is preset to - it doesn't make any difference to audio quality, in fact, even if I drop it to 1000000 it still sounds the same.

Any other suggestions? Current code below

```cpp
#include <WiFi.h>
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include "AudioFileSourceBuffer.h"
#include "driver/i2s.h"
#include "SD.h"
#include "SPI.h"
#include "FS.h"

//Initialize ESP8266 Audio Library classes
AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioFileSourceBuffer *buff;
AudioOutputI2S *out;

volatile bool playing = 0;
volatile byte loadTrack = 0;

uint8_t ledpin = 13;
bool ledphase = 0;
uint32_t ledms = millis();

void setup() {  
  WiFi.mode(WIFI_OFF);
  Serial.begin(115200);
  
  out = new AudioOutputI2S();
  out -> SetPinout(14,15,32);
  mp3 = new AudioGeneratorMP3();

  delay(1000);
  Serial.print("Initializing SD card...");
  if (!SD.begin(33,SPI,8000000))
  {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
  delay(1000);

  pinMode(ledpin, OUTPUT);
}

void loop() {
  // This is to check code is non-blocking:
  if(millis() - ledms > 1000) {
    ledms = millis();
    ledphase = !ledphase;
    digitalWrite(ledpin, ledphase);
  }
  
  if(!playing) {
    loadTrack = 1;
  }
  
  if(loadTrack) { // Load the track we want to play
    //Stop the current track if playing
    if(playing && mp3->isRunning()) mp3->stop();

    file = new AudioFileSourceSD("/Dover2.mp3");
    buff = new AudioFileSourceBuffer(file, 2048);
    
    out -> SetGain(0.16); //Set the volume
    mp3 -> begin(buff,out); //Start playing the track loaded
    playing = 1;
    loadTrack = 0;
  }

  if(playing && mp3->isRunning()) {
    if (!mp3->loop())
    {
      mp3->stop();
      playing = 0;
      Serial.println("Stopped");
    }
  }
}
```

imranaalam, 2020-04-02:
Have you solved the issue?
What is your schematic?

AidanTek, 2020-04-02:
> Have you solved the issue?
> What is your schematic?

Hiya, sorry this was many months ago - I moved on to some different hardware to realise the project in another way. I'm not going to pick this up any time soon sorry